PiSupply/PiJuice #408: Cannot select battery profile

kklmn, 2019-10-29:
Hello,

My battery profile is by default set as BP6X (this already contradicts the manual), I want to change it to BP7X but I can't.

Firstly, I don't see any 'Apply' button. If I just select BP7X from the list, all the parameters become empty. If I use 'Custom' box, the parameters appear (and are for BP6X) but whatever I change is ignored, i.e. I see the old values when I reopen the dialog.

Secondly, I see "Profile selected by: DIP_SWITCH". This differs from what I see in your manual. Where can I change it?

The DIP switch itself has no influence on the dialog. Maybe it does something for the actual charging parameters, I don't know it.

You say in the manual it is important to select the right profile. Then HOW?

ChristopherRush, 2019-10-29:
I'm assuming you are using the GUI, in which case just expand the window by
dragging it out then apply the changes.

kklmn, 2019-10-29:
ok, thanks for solving the quest! :)

closed 2019-10-29 by kklmn