PiSupply/PiJuice #409: Pi 4 does not reboot when powered through Pi

ChristopherRush, 2019-10-31:
Not sure if this is isolated or not so would be good if someone else can test this theory. 

I have the Pi 4 powered through its own USB C port with the PiJuice HAT connected. I have set `software halt power off` under `system task`. Once I have shutdown the Raspberry Pi through the Pi menu I then cannot boot it back up using a single press of the button on the PiJuice. I tested this on a 3B+ and it worked as intended.

tvoverbeek, 2019-10-31:
Works for me on a Pi4/PiJuice HAT powered via the Pi4 USB-C port.  Does reboot by SW1 button press after software shutdown.
But you do have to wait till the delay period in the 'software halt power off' has expired.

ChristopherRush, 2019-11-01:
It seems like GPIO power input to the PiJuice is not removed after 20 seconds (Delay period set). I have to remove the USB C power to the Pi 4 in order for it shutdown and only then can I restart using SW1.

ChristopherRush, 2019-11-01:
*Solved, with thanks to @tvoverbeek 
Solution was to update the bootloader with the latest release as per instructions here - https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

Release notes - https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md

closed 2019-11-01 by ChristopherRush