PiSupply/PiJuice #411: pijuice_cli = No PiJuice found or Failed to use I2C bus, + other weirdness

tomfanning, 2019-11-05:
When starting pijuice_cli I get "No PiJuice found or Failed to use I2C bus"

When running python3 /usr/bin/pijuice_cli.py the CLI starts but if I go into Status it crashes:

`Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2007, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1907, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 191, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 251, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 206, in get_status
    general_info += ", %3d°C, %s" % (temp, status['battery'])
KeyError: 'battery'`

However if I add `dtoverlay=i2c-bcm2708` to `/boot/config.txt` then `python3 /usr/bin/pijuice_cli.py` works. (bare `pijuice_cli` does not, same Failed to use I2C bus error as first case).

Pi 4B, Raspbian, latest apt updates, latest firmware through rpi-update. PiJuice firmware 1.3 (status: up to date).

`root@raspberrypi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --`

`root@raspberrypi:~# python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import Pijuice
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'Pijuice' from 'pijuice' (/usr/lib/python3/dist-packages/pijuice.py)`

ChristopherRush, 2019-11-06:
> File "", line 1, in

You are logged in as root, any particular reason for this? Try logging in
as user `pi` and then run pijuice_cli.

tomfanning, 2019-11-06:
That works, but why shouldn’t the root account be able to use the CLI?

tvoverbeek, 2019-11-06:
The pijuice_cli program is a wrapper which runs 'python3 /usr/bin.pijuice_cli.py' as user pijuice.
The wrapper will exit when it is called as root (security reasons).
When you run 'python3 /usr/bin/pijuice_cli.py' as root, it works but the pijuice service will not run any user functions from the root account

tomfanning, 2019-11-06:
Well that’s a pain, and also inconsistent with every other piece of
software on the Pi.

pedrosk, 2019-11-06:
I have a similar problem reproducible on PI2.  I run a headless Jessie and  I am ssh-ed ad me.

I do not have pi user for the very security reasons. 
To remove "pi" user is one of the first steps I do on my builds.

In my case I get this result when I enter certain menus. 

Is there a way to run it as non "pi" user or "root"?
what priviledges does this user need have and what groups it needs to be a member of?

tvoverbeek, 2019-11-06:
@pedrosk Your user (normal user) has to belong to the i2c and pijuice groups.
Since you are running headless, does pijuice_cli run with these settings?

tomfanning, 2019-11-06:
Yes that works, but it’s not the solution I want.

pedrosk, 2019-11-07:
In a way agree with @tomfanning  that the code should be able to run with any user or a group. 
I guess i2c group is an exception?

I hope this is not too generic or redundant, I am just really trying to understand.

so here on my system i2c group does exist, but not pi_juice. 

$ getent group i2c
i2c:x:998:
$ getent group pi_juice
returns nothing

to create a pi_juice group and to set this up for user to be a part of pi_juice group  should not be a problem. What special privileges should the pi_juice group have? (why is the pi_juice group significant?)

Look forward to hear back.

Thanks

pedrosk, 2019-11-07:
`lsmod
Module                  Size  Used by
i2c_bcm2708             5994  0 
binfmt_misc             7988  1 
ctr                     4263  4 
ccm                     9009  2 
arc4                    2211  2 
rt2800usb              18824  0 
rt2800lib              80070  1 rt2800usb
rt2x00usb              12798  1 rt2800usb
rt2x00lib              48694  3 rt2800lib,rt2800usb,rt2x00usb
mac80211              655480  3 rt2800lib,rt2x00lib,rt2x00usb
cfg80211              543027  2 rt2x00lib,mac80211
rfkill                 20851  2 cfg80211
crc_ccitt               1771  1 rt2800lib
snd_bcm2835            24427  0 
snd_pcm                98501  1 snd_bcm2835
snd_timer              23968  1 snd_pcm
i2c_bcm2835             7167  1 
snd                    70032  3 snd_timer,snd_bcm2835,snd_pcm
bcm2835_gpiomem         3940  0 
fixed                   3285  0 
uio_pdrv_genirq         3923  0 
uio                    10204  1 uio_pdrv_genirq
i2c_dev                 6913  2 
ipv6                  408971  31 
`

```
groups pedro
pedro : pedro sudo i2c pijuice_cli
```

Result if non root user: pedro
`pedro@pijuice:~$ pijuice_cli.py `

[image]

and I did enable  i2c and rebooted the PI2.

Still not working. What am I doing wrong?

tvoverbeek, 2019-11-07:
@pedrosk From your lsmod output you have both i2c_bcm2708 and i2c_bcm2835 loaded. Only one should be loaded. i2c_bcm2708 is only needed on Pi4.
Also the group name is not pijuice_cli, but pijuice.

pedrosk, 2019-11-07:
1) I did fix the module (modprobe -rv i2c_bcm2708)

>

>  pedro@pijuice:~$ lsmod
> Module                  Size  Used by
> binfmt_misc             7988  1 
> ctr                     4263  4 
> ccm                     9009  2 
> arc4                    2211  2 
> rt2800usb              18824  0 
> rt2800lib              80070  1 rt2800usb
> rt2x00usb              12798  1 rt2800usb
> rt2x00lib              48694  3 rt2800lib,rt2800usb,rt2x00usb
> mac80211              655480  3 rt2800lib,rt2x00lib,rt2x00usb
> cfg80211              543027  2 rt2x00lib,mac80211
> rfkill                 20851  2 cfg80211
> crc_ccitt               1771  1 rt2800lib
> snd_bcm2835            24427  0 
> snd_pcm                98501  1 snd_bcm2835
> snd_timer              23968  1 snd_pcm
> snd                    70032  3 snd_timer,snd_bcm2835,snd_pcm
> i2c_bcm2835             7167  1 
> bcm2835_gpiomem         3940  0 
> fixed                   3285  0 
> uio_pdrv_genirq         3923  0 
> uio                    10204  1 uio_pdrv_genirq
> i2c_dev                 6913  2 
> ipv6                  408971  33 

2) added **pijuice** group and added user to the group

```
pedro@pijuice:~$ groups pedro
pedro : pedro sudo i2c pijuice_cli pijuice
```

3) reboot

4) same issue....

any ideas what else am I doing wrong?

tvoverbeek, 2019-11-07:
First we have to get i2cdetect working.
I am assuming /dev/i2c-1 exists, since the i2c-dev module is loaded. Correct?
Can you check if the PiJuice is properly mounted on the 40 pin connector?
(have done this myself, one row off).
If not connected properly, that would explain the empty i2cdetect output.

pedrosk, 2019-11-07:
1) hardware connected correctly, in fact PI is powered by PIjuice heat (power USB connector in RASPI is not used).

2) Looks like owned by root and group i2c read and write.  Is this sufficient?

 ```
pedro@pijuice:~$ ls -la  /dev |grep -i i2c
crw-rw----  1 root i2c    89,   1 Nov  7 16:49 i2c-1
```

tvoverbeek, 2019-11-07:
@pedrosk That looks ok.
In your lsmod output there is also the bcm2835_gpiomem module.
For what reason?
As far as I know this module memory maps all gpio registers -> i2c drivers will not work.

pedrosk, 2019-11-07:
good question.

Not sure. I don't recall installing it.

```
sudo modprobe -rv bcm2835_gpiomem
rmmod bcm2835_gpiomem
```

not sure if this helps: 
```

sudo modprobe -rv bcm2835_gpiomem
rmmod bcm2835_gpiomem

 sudo  i2cdetect -l
i2c-1	i2c       	bcm2835 I2C adapter             	I2C adapter

```

Same result as above for both i2cdetect -y 1 and pijuice_cli.py

tvoverbeek, 2019-11-07:
I am out of suggestions.
Could you start from a clean raspbian-lite, install pijuice-base and first test with user pi.
Then add your user pedro, assign pedro to the correct groups, give sudo permissions and then try with user pedro.
It seems you have a lot of modules for which you do not have any use.

PS Hope you are not using a hyperpixel screen, since this uses all gpios, also the i2c ones.

pedrosk, 2019-11-07:
I was just thinking the same. Will give that a shot as soon as I find a few cycles I can dedicate to this. 
It probably will be different since it will newer kernel and built.

Current: 
```
pedro@pijuice:~$ uname -a
Linux pijuice 4.9.35-v7+ #101 4 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux ```

Really appreciate all the help.

tvoverbeek, 2019-11-07:
Yes, 4.9.35 is pretty old (> 2 years) and still Wheezy (?).
Current pijuice packages are on Stretch and Buster and require python3.
Curremt kernel is 4.19.75

tomfanning, 2019-11-08:
Just a note, while I hope you get sorted, this is no longer the original
issue I reported, which I still consider outstanding (root user can’t use
pijuice_cli).

pedrosk, 2019-11-09:
I am starting to wonder if there is something wrong with the device itself. 

here is what I did:
I took a lots of steps back, and followed instructions as closely as possible. tested on both PI2 and PI4.

NOTE: each test was done on a fresh install!

run as user "pi" as recommended by the instructions. 

1) Downloaded new raspian headless image from and imaged a the chips. 
`2248146944 Nov  7 16:28 2019-09-26-raspbian-buster-lite.img`

After install you can see that this is the newest raspian version: (run apt-get update and upgrade)

```
pi@pijuice:/usr/bin $ uname -a
Linux pijuice 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
pi@pijuice:/usr/bin $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@pijuice:/usr/bin $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

than made sure I am running python3:
```
pi@pijuice:~ $ python --version
Python 3.7.3
pi@pijuice:~ $ which python
/usr/bin/python
```
next installed the cli version of pijuice:
`sudo apt-get install pijuice-base`

Checked lsmod: (this is from PI4)
pi@pijuice:~ $ lsmod

> Module                  Size  Used by
> bnep                   20480  2
> hci_uart               40960  1
> btbcm                  16384  1 hci_uart
> serdev                 20480  1 hci_uart
> bluetooth             389120  24 hci_uart,bnep,btbcm
> ecdh_generic           28672  1 bluetooth
> 8021q                  32768  0
> garp                   16384  1 8021q
> stp                    16384  1 garp
> llc                    16384  2 garp,stp
> rtc_ds1307             24576  1
> vc4                   176128  0
> v3d                    61440  0
> gpu_sched              28672  1 v3d
> raspberrypi_hwmon      16384  0
> hwmon                  16384  2 rtc_ds1307,raspberrypi_hwmon
> brcmfmac              311296  0
> drm_kms_helper        184320  1 vc4
> brcmutil               16384  1 brcmfmac
> sha256_generic         20480  0
[... 116 lines cut ...]
>   File "/usr/bin/pijuice_cli.py", line 191, in __init__
>     self.main()
>   File "/usr/bin/pijuice_cli.py", line 251, in main
>     status_args = self.get_status()
>   File "/usr/bin/pijuice_cli.py", line 206, in get_status
>     general_info += ", %3d°C, %s" % (temp, status['battery'])
> KeyError: 'battery'

                  
Firmware claims firmware: 1.1 and firmware 1.3 available:

[image]

General says its unable to connect to to device:
[image]

In summary, I am not sure if I am missing a step ... or something else. I took two days to test scenarious and different boards,  but this reproduces same on both PI2 and PI4. 

Not sure if it means anything, but I was one of the first kickstarter supporters and this is one of the first boards. I did not get to this project till now, but I was very excited about and I really would love to make it work.

Do you have any other ideas, what may need to be done to make this work? Is there a chance that this is just a not working well because of CLI (which would be very odd), or maybe because I am using Raspian (debian) instead of fully loaded Noob?  I was also hoping to use it on other boards like Pine and Rock64 based boards ... :( But I am not sure if that is possible either. 

Any advice you may have?

Thanks.

ChristopherRush, 2019-11-09:
Why haven't you updated the Firmware?

pedrosk, 2019-11-09:
@ChristopherRush is the old firmware the cause of the issues I am seeing?

ChristopherRush, 2019-11-09:
@pedrosk I couldn't say for sure but I would highly recommend you update it since it has fixed a number of issues we had in the past.

pedrosk, 2019-11-09:
@ChristopherRush Thanks. Makes sense. Will do and I hope it will help.

From the details I provided is there anything you think I omitted and did incorrectly that could lead to this behavior?

tvoverbeek, 2019-11-10:
@pedrosk Just a weird idea. Since your reported firmware version is 1.1 I am wondering if this is
an original kickstarter pijuice which was never used until now.
I wonder if you ever removed the little plastic tab between the BP7X battery and the battery terminal. Could have some weird effects since no battery is detected.
Anyway, your i2cdetect output looks OK. The 'UU' at 0x68 means the device-tree fragment from the eeprom was read and the rtc initialized. So i2c communication with the pijuice is working.
Note: for pi4 you need to use the old i2c driver. In /boot/config.txt add the line `dtoverlay=i2c-bcm2708` in the `[pi4]` section.
Your lsmod output shows the i2c-bcm2835 driver.

pedrosk, 2019-11-10:
@tvoverbeek, @ChristopherRush the tab was pulled out from the beginning (sitting here on my messy table), but I did wiggle the battery and I could see the spring mechanism connecting to the battery. I believe the battery was connected OK, because I did run it off the battery while testing the external industrial outdoor antennae with PI2. But maybe it was not connected too well? 

However,  it looks like this:  `dtoverlay=i2c-bcm2708 in the [pi4] section.` did the trick.

[image]

Can you provide guidance on this setting for the other older PIs with the newest Raspian version (for completeness)? and in fact why does this need to be done separately after enabling i2c with raspi-config? 

(will try to upgrade the firmware later today) 

And thank you sooooo much for the dedication and help thus far.

tvoverbeek, 2019-11-10:
@pedrosk The `dtoverlay=i2c-bcm2708` is only needed for the Pi4. All the older Pi models work fine with the default i2c-bcm2835 module.
In the next version of the firmware this workaround for the Pi4 is not needed any more.

pedrosk, 2019-11-10:
@tvoverbeek I did only enable it via raspi-config on the old PI models.
 Does this setting need to be manually added to the boot config or does enabling i2c via raspi-config is sufficient?

tvoverbeek, 2019-11-10:
@tomfanning Found out why pijuice_cli does not work for root.
The setuid wrapper (in case of root) drops privileges by setting the real and effective userid to pijuice. Same for the real and effective groupid.
However the supplementary groups list is not changed. Since the supplementary groups list for root does not include the i2c group, pijuice_cli is executing the python3 interpreter as user pijuice without access to the i2c bus.
When doing this from e.g. the pi user the supplementary groups list does include i2c and therefore it works.
Working now on modifying the setuid wrapper so in case it is called from root the supplementary groups list is updated to include the i2c group.
In the future you should be able to run pijuice_cli (and pijuice_gui) from the root account.

tvoverbeek, 2019-11-10:
@pedrosk Enabling i2c via raspi-config is sufficient for the older Pi models.
Note the install of the pijuice-base package does this for you.

pedrosk, 2019-11-11:
@tvoverbeek @ChristopherRush 
I am still puzzled. This works on PI4 but not on PI2:
```
 grep pijuice /etc/group
i2c:x:998:pi,pijuice
pijuice:x:1001:pi
```

_NO ports seems to used_

>  i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 70: -- -- -- -- -- -- -- -- 

> 
> lsmod
> Module                  Size  Used by
> ctr                    16384  4
> aes_arm_bs             20480  0
> crypto_simd            16384  1 aes_arm_bs
> cryptd                 24576  1 crypto_simd
> ccm                    20480  6
> 8021q                  32768  0
> garp                   16384  1 8021q
> stp                    16384  1 garp
> llc                    16384  2 garp,stp
> arc4                   16384  2
> rt2800usb              28672  0
> rt2800lib             102400  1 rt2800usb
> rt2x00usb              24576  1 rt2800usb
> rt2x00lib              57344  3 rt2800lib,rt2800usb,rt2x00usb
> mac80211              692224  3 rt2800lib,rt2x00lib,rt2x00usb
> sha256_generic         20480  0
> cfg80211              614400  2 rt2x00lib,mac80211
> rfkill                 28672  3 cfg80211
> crc_ccitt              16384  1 rt2800lib
> snd_bcm2835            24576  1
> raspberrypi_hwmon      16384  0
> hwmon                  16384  1 raspberrypi_hwmon
> snd_pcm               102400  1 snd_bcm2835
> snd_timer              32768  1 snd_pcm
> bcm2835_codec          36864  0
> bcm2835_v4l2           45056  0
> v4l2_mem2mem           24576  1 bcm2835_codec
> snd                    73728  5 snd_timer,snd_bcm2835,snd_pcm
> bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
> videobuf2_dma_contig    20480  1 bcm2835_codec
> v4l2_common            16384  1 bcm2835_v4l2
> videobuf2_vmalloc      16384  1 bcm2835_v4l2
> videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
> videobuf2_v4l2         24576  3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
> i2c_bcm2835            16384  0
> videobuf2_common       45056  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
> videodev              200704  6 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
> media                  36864  3 bcm2835_codec,videodev,v4l2_mem2mem
> vc_sm_cma              36864  1 bcm2835_mmal_vchiq
> uio_pdrv_genirq        16384  0
> fixed                  16384  0
> uio                    20480  1 uio_pdrv_genirq
> i2c_dev                20480  0
> ip_tables              24576  0
> x_tables               32768  1 ip_tables
> ipv6                  450560  24
> 

I have a batch of new PI4s (they finally arrived after a few month), but I also have a large number of PI2, PI3 and even old original PIs in production for years that I was hoping to use with pijuice. 

Did something change in the code over the time that this seems not work? 
It probably is not worth the time or effort to you guys or me to decouple this, but I still find it quite interesting .... I was really debating if I should even write back to this thread since it is working on PI4 ... 
Any clues what might have changed over time?

tvoverbeek, 2019-11-11:
@pedrosk First thing to do is to get i2cdetect reporting the device.
All necessary modules seem to be present. You have a lot of video modules. Anything connected which uses GPIO2 and 3 (pins 3 and 5) and therefore blocks i2c?

pedrosk, 2019-11-11:
@tvoverbeek 
Its a fresh headless install with latest OS, the only thing that is used is a WiFI via USB as seen the pictures. 

Pi2 v 1.1 does not has a wifi on board, plus this is intended to work outside at some point and would be in a outdoors box impacting the onboard wifi regardless, which would dictate for an external antenae.  Maybe I am wrong, but this  USB should be blocking it ... or?  What do you think?

[image]
[image]
[image]

Besides the USB wifi nothing else was ever connected to this PI2 board. So nothing should be listening to GPIO (the same settings work on PI4, in fact I pulled the chip and put to PI4 and booted and worked after adding the setting for the boot config )

PS: The firmaware is at newest (1.3)

tvoverbeek, 2019-11-11:
Can you run the following command: `raspi-gpio get | head`?
Here is my output
```
pi@raspberrypi:~ $ raspi-gpio get|head
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=0 func=INPUT pull=UP
GPIO 1: level=1 fsel=0 func=INPUT pull=UP
GPIO 2: level=1 fsel=4 alt=0 func=SDA1 pull=UP     <----
GPIO 3: level=1 fsel=4 alt=0 func=SCL1 pull=UP     <----
GPIO 4: level=1 fsel=0 func=INPUT pull=UP
GPIO 5: level=1 fsel=0 func=INPUT pull=UP
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 7: level=1 fsel=0 func=INPUT pull=UP
GPIO 8: level=1 fsel=0 func=INPUT pull=UP
pi@raspberrypi:~ $ 
```
GPIO2 and GPIO3 should show alt=0 and the i2c funcs SDA1 and SCL1

pedrosk, 2019-11-11:
pi@pijuice:~ $ raspi-gpio get | head
BANK0 (GPIO 0 to 27):
GPIO 0: level=0 fsel=0 func=INPUT
GPIO 1: level=0 fsel=0 func=INPUT
GPIO 2: level=0 fsel=4 alt=0 func=SDA1
GPIO 3: level=0 fsel=4 alt=0 func=SCL1
GPIO 4: level=0 fsel=0 func=INPUT
GPIO 5: level=1 fsel=0 func=INPUT
GPIO 6: level=1 fsel=0 func=INPUT
GPIO 7: level=1 fsel=0 func=INPUT
GPIO 8: level=1 fsel=0 func=INPUT

tvoverbeek, 2019-11-11:
Your output does not show the pull=UP for GPIO2 and 3. Is this an older version of raspi-gpio or do you have all pull-ups and pull-downs disabled? The I2C bus needs pull-ups

pedrosk, 2019-11-12:
Like I mentioned its raspi PI2 v 1.1 according the [ specification ](https://en.wikipedia.org/wiki/Raspberry_Pi) section these were made between August 2015 and October 2016. How can I get it pulled up? My search of internet did not come back with anything that made any sense. If you have a good link to read I would love to get educated.  From what I read there are internal resitors to pull-up and pull-dow, so I imagine that there is some kind of software solution to this.

Appreciate and look forward to learn.

tvoverbeek, 2019-11-12:
@pedrosk One of my RPis is also a Pi2 V1.1. I put the SD-card from my Pi4+PiJuice in the Pi2 and moved the PiJuice to the Pi2. My Pi2 has an Edimax USB WiFi dongle.
The raspi-gpio output on my Pi2 is identical to yours (i.e. no pull-up/down information on the Pi2).
I2C works and I can use pijuice_cli and pijuice_gui without problems.
Do you have an other I2C device which you could use to test if i2c is working?
There is a chance the I2C GPIO pins are fried by applying 5V to them.
I guess the only other thing to try is to start from scratch with a fresh install of Buster on the Pi2
and see if the i2c is working.

pedrosk, 2019-11-14:
I am afraid you are correct.
 I managed to test the same deplyment on PI3 and worked and today I found a version PI2 1.1 from 2014 identical to the old one and pijuice worked well.

Appreciate all the testing. Since nothing was ever connected to the GPIO  bus I did not consider that option, so I am not sure now it got burned out.

Again, appreciate

JohnStarich, 2020-01-03:
I ended up with the same stack trace on my Pi4 as the OP, and the same solution worked for me:
>In /boot/config.txt add the line `dtoverlay=i2c-bcm2708` in the `[pi4]` section.
Big thank you for that! 👍 

Is there somewhere I can track when the new firmware will be released? (Best I found was the commit log on the Firmware directory, but the mentioned 1.5 version is not yet available in the gui/cli.)