PiSupply/PiJuice #413: Battery failing

smcavoy, 2019-11-19:
The battery that came with the PiJuice is failing. When power is removed the unit stays powered for a few seconds then the unit powers off. 

I have tried another battery attached to the terminal connections and the unit stays powered for sometime. 

The voltage on the battery that came with PiJuice shows 4.1 volts. The unit was purchased under a year ago.

Any suggestions?

uGiFarukh, 2020-01-13:
Is it possible for you to test the battery manually and see if the battery can hold its voltage accordingly? Check for the full, nominal and drained voltages of the battery. Try testing the battery manually using some sort of resistive load. If the battery functions as expected, something might went wrong with the battery connector on the pijuice board. BTW which battery profile is set on the pijuice software? And I am assuming the battery you got with the PiJuice is BP7X.

RumoOr, 2020-02-08:
Same Issue for me. PiJuice dosnt react after Micro USB connection directly to the PiJuice. Sometimes when i take out the battery and put it in again, it works for the next SW1 start. but after some minutes or after Pi shutdown (SW1 10s) the PiJuice doesnt show any LED. Pressing SW1 again leads to no reaction again. I Use the shipped BP7X 1820 mAh 4.2V battery. DIP Switch is setup correctly with 1=On;2=Off.
Not sure if this issue depends on battery, battery contact or the MCU. Any suggestions?

tvoverbeek, 2020-02-08:
Smells like the BP7X is at end of life. Reduced capacity so it shuts down immediately when loaded.
My own 2 BP7X after intensive use show the same behavior. They are basically unusable now.
Note they were originally used for the Motorola Droid which is more than 8 years ago. Unknown how long they have been on the shelf or even if they were used before and resold.
Suggest to raise a support ticket with Pi-Supply

RumoOr, 2020-03-01:
Recently noticed strange behaviour. see images below.

Pijuice leds not working. Battery status not available.
[Image 1](https://i.imgur.com/yW1TM0t.jpg)

When i put my finger on the board (witout any pressure), leds shows status:
[Image 2](https://i.imgur.com/Tfizoe4.jpg)

broken pijuice board?

closed 2021-11-21 by smcavoy