PiSupply/PiJuice #415: Using SDA and SCL pins causing communications failure

JVTEAM, 2019-11-22:
I'm looking to use the gpio 2 and 3 pin to power a small LCD screen.  However, when I do this the board no longer communicates to the battery.  How can I fix this?

tvoverbeek, 2019-11-22:
GPIO 2 and 3 are used for I2C communication. Communication with the PiJuice uses I2C, hence it does not work when you use  GPIO 2 and 3 for something else.
Note: If your display uses DPI (e.g. Pimoronis hyperpixel displays) the PiJuice will not work. These displays claim almost all GPIOs.

JVTEAM, 2019-11-22:
are there any alternatives to use an i2c lcd screen when the pijuice is attached?

tvoverbeek, 2019-11-23:
If there are no address conflicts on the I2C bus it should work.
PiJuice uses addresses 0x14 and 0x68. What address is your LCD screen using?
if your Pi is a Pi4 then you should use the old i2c driver by adding the line `dtoverlay=i2c-bcm2708` to /boot/config.txt. On the Pi4 the PiJuice has communication problems when using the default i2c driver.

JVTEAM, 2019-11-23:
This is a pi zero running buster.

The default value of I2C address is 0x3c.

tvoverbeek, 2019-11-23:
Does the PiJuice work without the display connected?
Does the display work without the PiJuice mounted on the Pi-Zero?

When you use both I assume the PiJuice sits on the 40 pin GPIO connector of the Pi-Zero and
you use the 4 wire bundle to connect the display to the 40 pins on the PiJuice.
Maybe the extra cables make the I2C signals bad. You could then try to lower the I2C clock speed.
Look at i2c_arm_baudrate in /boot/overlays/README.

JVTEAM, 2019-11-24:
Yes.  It works when the screen is not attached.  As soon as I plug in the i2c cords...comm failure.

I used a pin extender to make the pins go through the pi juice then attach the jumper wires.

tvoverbeek, 2019-11-24:
Since you use a pin extender, I assume you have a PiJuice-Zero not the PiJuice HAT.
Repeating my question: Does the screen work when attached to the Pi-Zero without the PiJuice-Zero?

JVTEAM, 2019-11-24:
yes, its a pi zero.  And also yes, the screen works without the pijuice attached.

JVTEAM, 2019-11-25:
i2c_arm_baudrate is (default "100000")

How do you adjust the baudrate and what do you recommend to try?

tvoverbeek, 2019-11-25:
Since I2C is enabled the line `dtparam=i2c_arm=on` should already be in /boot/config.txt.
You can add a baudrate parameter, i.e. `dtparam=i2c_arm=on,i2c_arm_baudrate=50000`.
This will halve the baudrate to 50 kHz.

An other problem could be that your display driver software changes the baudrate to 400000 (fast I2C). The PiJuice-Zero cannot handle an I2c baudrate of 400kHz.

shawaj, 2020-10-24:
@JVTEAM did you get this working?

shawaj, 2020-12-05:
Closing as no reply in over a month from @JVTEAM 

Reply if you disagree

closed 2020-12-05 by shawaj