PiSupply/PiJuice #417: Button events - inconsistent statuses returned

emil2099, 2019-11-26:
I am trying to get button events through Python, but all buttons return different results on single press:
`'SW1': 'NO_EVENT'`
`'SW2': 'SINGLE_PRESS'` - this is expected behaviour to me
`'SW3'; 'PRESS' `followed by `'RELEASE'`

Is this expected and is there a way to make buttons respond in the same way, and in particular get SW1 return something?

I run PiJuice with default configuration on Raspberry Pi 3 and tested using the following code:

```
from pijuice import PiJuice
import time

pijuice = PiJuice(1, 0x14)

while True:
    time.sleep(0.5)
    print(pijuice.status.GetButtonEvents())
```

tvoverbeek, 2019-11-26:
The physical button presses are processed by the MCU and then reported to the RPi on request.
Look st the description in the sofftware README: https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-config-buttons-menu
With the default assignments you get the behavior you are seeing. Note that when the pijuice service is running it will eat some of the events and interfere with your sample program.

emil2099, 2019-12-03:
@tvoverbeek That solved it - thanks! 

Another slight observation is that the Apply button on the above mentioned screen is actually hidden in the GUI, so need to extend the window in order to see it. Took a couple of failed attempts to figure that one out 😁

closed 2019-12-03 by emil2099

tvoverbeek, 2019-12-03:
@emil2099 The GUI not showing the Apply button is a known problem (caused by the change to Buster).
It is already fixed in Github, so if you want you can replace pijuice_gui.py with the current version on github