PiSupply/PiJuice #418: Is it possible to power on a RPI (model 3+) when it is powered up via micro usb on the PiJuice?

Nelvalhil, 2019-11-27:
Just like a RPI would do without the PiJuice. So basically it starting up without having to use the physical buttons on the PiJuice? 

Thanks in advance :)

ChristopherRush, 2019-11-27:
Yes it is. Simply press the SW1 button on the PiJuice HAT or configure
wakeup on charge in the PiJuice software under `system task`.

Nelvalhil, 2019-11-27:
> Yes it is. Simply press the SW1 button on the PiJuice HAT or configure wakeup on charge in the PiJuice software under `system task`.
> […](#)

I will check it out, thank you very much

Nelvalhil, 2019-11-30:
> Yes it is. Simply press the SW1 button on the PiJuice HAT or configure wakeup on charge in the PiJuice software under `system task`.
> […](#)

Hi, could you please answer me this, too? I'm looking to clone the sd card in the RPI, for more projects just like what I am working on right now. It will have a PiJuice, too. Are settings for the PiJuice stored on the RPI's SD card or does it has its own memory?

mchollett, 2019-12-06:
I thought I had my PiJuice properly setup and working but I'm having problems.  I want to use it as a UPS for the RPi.  When the power is shutoff, wait a period of time and then properly shutdown the RPi.  When power is supplied again, I want the unit to power back up.  If I start it and then turn it off it's usually reliable.  If it sits overnight, it doesn't reliably start back up.  The battery doesn't seem to be draining overnight as I get a blue charge indicator when I power up first thing in the morning.

I have power supplied to the PiJuice via micro usb.

I have wake on charge set with a value of 1 but it doesn't reliably restart.  I have shutdown and poweroff set but the delay doesn't seem to make much difference.  I change from 100 to 300 and I don't see a delay change.  Is this seconds or milliseconds ?

I'm using the CLI and cannot use the GUI.

I think this is a popular setup, maybe you could do a setup sheet on this ?  Clearly show how to setup this operation in both the CLI and GUI.

What should I try to get the restart working reliably ?

Thanks,
Mark

Nelvalhil, 2019-12-10:
> I thought I had my PiJuice properly setup and working but I'm having problems. I want to use it as a UPS for the RPi. When the power is shutoff, wait a period of time and then properly shutdown the [...]

Hi Mark, I've looked into your issue and I think this little guide (from this [topic)](urlhttps://github.com/PiSupply/PiJuice/issues/160) would work for you 

Started with power connected to the PiJuice USB input.
Setting 'No Power' in System Events to SYS_FUNC_HALT_POWER_OFF
Setting 'Wakeup on charge' enabled in System Task with trigger level 0%.
(System Task should also be enabled)
When I remove the power from the PiJuice USB input it takes a few seconds before LED D2 starts blinking red (=sending the 'sudo halt' command to the Pi), the Pi shuts down and about a minute later the red LED on the Pi goes off (=Power off for the Pi)
Then the Pijuice goes to low power mode: slow low intensity blink of LED D1.
When I then reconnect the power to the PiJuice USB input, the Pi boots immediately.

Let me know how it goes

mchollett, 2019-12-11:
Tried this and it sort of works.
Power on unit with button
Remove power and unit shuts down properly
Wait some period of time - maybe 10 minutes
Apply power to the Pijuice
Everything starts back as it should.
However.........
If I remove power and let it sit overnight
Apply power to the Pijuice
Blue light comes on PiJuice but Raspberry Pi does not start up
I have to use the switch to start.

Interesting note...........
I have an older PiJuice that works fine like this.
I even took the JSON config file off that unit, put it on the new unit, changed the battery type to be correct, and it does not start up properly.

Was something changed in the firmware between the old units and newer units (post long supply delay)?

What could be set that would prevent the startup if the wake on charge is set to 0 ?

Thanks,
mark

ChristopherRush, 2019-12-13:
I have been unable to replicate this issue you are having. I left my PJ over night and when applying power this morning it booted straight away as expected. I'm using Firmware v1.3. Are you able to send me your JSON file?

mchollett, 2019-12-17:
@ChristopherRush 
Were you able to take a look at the JSON and screenshots I posted ?

Thanks,
Mark

tvoverbeek, 2020-03-04:
@mchollett I know this thread is already more than 2 months old.
Your settings and JSON are fine.
When you let it sit overnight is the battery cold?
At low temperatures the battery will have reduced capacity and maybe not enough power to turn on the Pi.
The charging will warm up the battery and then the power-on by switch works.
Have you tried with the same battery in your older PiJuice?

Just as a side note, my own BP7X batteries are at end-of-life. Any load and the Pi goes off.

shawaj, 2020-10-24:
@Nelvalhil @mchollett do either of you need any further assistance or can we close this issue?

shawaj, 2020-12-05:
Not heard back from @Nelvalhil or @mchollett in over a month so closing now.

Reply if you disagree

closed 2020-12-05 by shawaj