PiSupply/PiJuice #421: EEPROM address 52 passthrough GPIO disconnects PiJuice #126

enriqueescobar-askida, 2019-12-05:
I put EEPROM Write Unprotected then EEPROM address to 52
to let RPi access AstroPi Stacked on top of PiJuice.
Unfortunately PiJuice cannot longer charge using GPIO nor microUSB.

How can I let RPi access AstroPi stacked on top of PiJuice, passing through PiJuice
And RPi still feeded by PiJuice?
---------------------- #126 related ---------------------
There is already a solved issue here #126 

You need to change the ID EEPROM address to 52.

Make sure that ID EEPROM Write unprotect is checked before changing the address to 52.

tvoverbeek, 2019-12-05:
@enriqueescobar-askida How do you know it does not charge? Is LED D1 blinking?
Which Pi  version are you using? Pi4 has problems communicating with the PiJuice in its default configuration.
As reported in issue #126 the Sense Hat (=Astro Pi board) is working together with the PiJuice.
Or do you have a different board (not Sense Hat)?

enriqueescobar-askida, 2019-12-12:
RPi 3 mondel B+ but it is ok after rebooting 👍 thanks

closed 2019-12-12 by enriqueescobar-askida