PiSupply/PiJuice #422: pijuice minutes-period always active

regie-techno, 2019-12-05:
Here what it is happening !
Each time I set Wakeup Enabled to true, I also check the checkbox "Minutes".
I press "set alarm", I press "back", and when I come back in the menu "Wakeup Alarm", the flag Minutes is set to false and the flag "Minutes period" is checked. 
I do not understand ! Is it a bug from pijuice card ?
Is there a way to solve the problem ?

Thanks a lot

tvoverbeek, 2019-12-05:
Drag the bottom window border down so the 'Apply" button becomes visible.
You have to click 'Apply' to make the changes effective.
The window size was OK on previous Raspbian versions, but there were changes in Buster which made the window too small. This is already fixed  on Github and will be in the next release.

regie-techno, 2019-12-05:
Thanks ! But I use pijuice_cli and the gui ...

tvoverbeek, 2019-12-05:
For pijuice_cli you have to select  <Set alarm > and press Enter to apply the changes.
You should then see "Wakeup alarm has been set successfully".

regie-techno, 2019-12-05:
I do this operation and I see "Wakeup alarm has been set successfully".
Then I press Back to go the main page. And when I go back to "Wakeup Alarm", the checkbox for Minutes is unchecked and the checkbox for 'Minutes period" is checked again ... I do not understand !!!

tvoverbeek, 2019-12-05:
@regie-techno I don't either. It does not happen to me. Do you have anything else running which could set the alarm?

regie-techno, 2019-12-05:
Anything ! I checked ! I am going to try to re-install pijuice-base

regie-techno, 2019-12-05:
SUCCESS ! It seems quite strange ever !
While I was testing this function, minutes and seconds were set to 0. If I set seconds to 1 (or any number except 0), it works fine !
I think obviously there is a bug here !

closed 2019-12-05 by regie-techno