PiSupply/PiJuice #423: PJ 2500mAh profile missing

ryanteck, 2019-12-07:
The battery profile for the 2,500mAh battery (https://uk.pi-supply.com/products/pijuice-2500mah-battery?_pos=1&_sid=328604b51&_ss=r) that we sell seems to be missing, I get the following profiles show.

[image]

shawaj, 2019-12-07:
@ryanteck I think this is because when we set up those profiles we didn't have that battery yet. Good spot!

shawaj, 2020-02-22:
I think we are also missing the 1200 and 1000 for the PiJuice HAT, although perhaps they are similar enough to the PJ Zero ones to be interchangeable.

The only real difference is the dimensions.

shawaj, 2020-12-24:
This will be fixed by #612