PiSupply/PiJuice #424: GUI is not opening

praveen7760, 2019-12-09:
i installed gui by using manual methods and i got the option , but when i try to open the software is is giving error, like
Failed to execute child process "/usr/bin/pijuice_gui" (No such file or directory)
but i checked , the file is there, also i tried cli, but it is also giving communication error.
please solve this issue.

ChristopherRush, 2019-12-09:
What OS are you using?

praveen7760, 2019-12-09:
debian LXDE 9.11 in rockpi board

praveen7760, 2019-12-09:
linaro@linaro-alip:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.11 (stretch)
Release:	9.11
Codename:	stretch

tvoverbeek, 2019-12-09:
@praveen7760 You wrote "Failed to execute child process "/usr/bin/pijuice_gui" (No such file or directory)"
There should be 2 files: /usr/bin/pijuice_gui and /usr/bin/pijuice_gui.py.
/usr/bin/pijuice_gui is a wrapper which executes 'python3 /usr/bin/pijuice_gui.py' as user pijuice.
Does 'sudo -u pijuice python3 /usr/bin/pijuice_gui.py' work?

praveen7760, 2019-12-10:
sudo -u pijuice python3 /usr/bin/pijuice_gui.py
command is working fine, but it is giving error saying that , communication error

praveen7760, 2019-12-12:
@tvoverbeek @ChristopherRush any update on my issue

tvoverbeek, 2019-12-12:
@praveen7760 Sorry, no update.
I am not familiar with the various Rock Pi boards. They use a different SoC. E.g. when they are 64-bit the wrapper programs for pijuice_cli and pijuice_gui will not work, since they are 32-bit only. You would have to rebuild them for 64-bit.
Also how I2C is handled may be different. You would have to study the installation scripts on Github and see how to do this on your Rock Pi. 
FYI On the Pi4 under Buster the default I2C driver does not work for the PiJuice (communication error) and there you have to use the previous driver. Do not know how the modules are organized in the Rock Pi version.

shawaj, 2019-12-12:
@praveen7760 @ChristopherRush @tvoverbeek - John has just informed me that we apparently have got a discussion with RockPi underway where they are interested to build support for some of our products.

Probably not enough demand for us to justify the time doing this ourselves, although I believe @ChristopherRush will be getting a RockPi soon so we can help with debugging but can't promise any support or any timelines or anything as it is not a priority at the moment.

praveen7760, 2019-12-13:
@shawaj  oh great, any new update please let me know ASAP

hipboi, 2019-12-14:
This is tom from Radxa, the team behind ROCK Pi. From what I can see on the hardware side, I think we only need the I2C on PIN3 and PIN5. The i2c bus number on ROCK Pi and Raspberry Pi are different. Is that configurable for the Juice Pi? Also, @shawaj  Could you check the low level cmd line interface?

tvoverbeek, 2019-12-14:
@hipboi You are right, for communication only I2C is needed via  GPIO connector pins 3 (SDA) and pin 5 (SCL).
For the low level base module (pijuice.py) the bus and address are configurable.
Typical python sequence is:
```
from pijuice import PiJuice
pj=PiJuice(<busno>,<address>)
```
For the RPi this is bus 1 and address 0x14.
However the other software (pijuice_cli.py, pijuice_gui.py, pijuice_sys.py and pijuice_tray.py
all use hard coded busno and address.
All need only a simple change  to specify an other bus than 1.

Let me know if you need more  info

hipboi, 2019-12-16:
@praveen7760  make sure you enabled the i2c7 on the ROCK Pi 4, either by editing the device tree or simply edit the `/boot/hw_intfc.conf` file to set `intfc:i2c7=on`. Then you can use i2c-tools to detect if the juice I2C device can be scanned. If you can find it at address 0x14, you can modify the python code and test the cmdline tools.

praveen7760, 2019-12-16:
ok, i will do that

praveen7760, 2019-12-16:
@hipboi i changed the config file , also found the bus no =0 for i2c7 and address, i changed the bus no in pijuice_gui.py , pijuice_cli.py
and executed "sudo -u pijuice python3 /usr/bin/pijuice_gui.py" commond ..still it is giving communication error.

akgnah, 2019-12-16:
This is akgnah from Radxa, the team behind ROCK Pi.

The problem arises because your system is 64-bit and pijuice_gui is 32-bit. These are the steps to make a 32-bit program run on a 64-bit system:

> dpkg --add-architecture armhf
apt-get update
apt-get install libc6:armhf libstdc++6:armhf

@praveen7760 Please try it.

Here are some results of executing i2cdetect after set intfc:i2c7=on:

> linaro@linaro-alip~ $ sudo i2cdetect -l
i2c-0	i2c       	rk3x-i2c                        	I2C adapter
i2c-1	i2c       	rk3x-i2c                        	I2C adapter
i2c-7	i2c       	rk3x-i2c                        	I2C adapter
i2c-9	i2c       	DesignWare HDMI                 	I2C adapter
linaro@linaro-alip~ 
linaro@linaro-alip~ $ sudo i2cdetect -y -a 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

praveen7760, 2019-12-17:
@akgnah i executed all the commnads you have provided, and
also found bus number :
linaro@linaro-alip:~$ mraa-i2c list
Bus   0: id=07 type=linux  default
Bus   1: id=02 type=linux 
Bus   2: id=06 type=linux 
i executed sudo i2cdetect -y -a 7 and  i got 
linaro@linaro-alip:~$ sudo i2cdetect -y -a 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
and i changed the bus number from 1 to 0 in all the python files
pijuice =PiJuice(0, 0x14)
after this i restarted the board for my reference , if i execute the pijuice_gui.py file ...still it is 
giving communication error.

@akgnah  can you provide the complete step to run the pijuice gui.

tvoverbeek, 2019-12-17:
@praveen7760 In i2cdetect you used bus 7, so in the python files you also have to use bus 7. I do not know if you can access i2c as normal user or if you have to be root (you used sudo with i2cdetect). If not possible as normal user then you will have to run the gui also as root (sudo).

praveen7760, 2019-12-18:
@tvoverbeek , @hipboi thanks , after i changed the bus number to 7 in all the python files , it is working fine.

tvoverbeek, 2019-12-30:
Closing the issue

closed 2019-12-30 by tvoverbeek