PiSupply/PiJuice #425: PiJuice Zero - adding a hard-reset switch?

GregHibbert, 2019-12-11:
The PiJuice Zero is a great little unit. I have been using the PiJuice Zero for a project with a Pi 3b running Buster, and have found that after some time (and dozens of power cycles), the Juice will get into a “confused” state. 

When it's "confused", it is effectively non-functional and causes the whole system to fail. The "confused" state includes inconsistent behaviour of SW2 (used as a power/shutdown/hard shutdown switch), inability to connect to the PiJuice via the Python API, and general weirdness including an RTC issue.

My power cycle (Pi boot) includes loading kernal drivers for the RTC and setting the system time from the RTC. This may or may not be causing the "confused" state, but an early symptom is seeing this repeated every second in the syslog:
Dec 11 08:37:37 raspberrypi kernel: [  218.094906] rtc-ds1307 3-0068: write error -5
Dec 11 08:37:39 raspberrypi kernel: [  220.095013] i2c i2c-3: sendbytes: NAK bailout.

(Note that I set the RTC driver using modprobe; I'm running the PiJuice over I2C3 provided by the HyperPixel 4 touch-screen display and the automatic loading doesn't seem to be an option.
> echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-3/new_device)

So far the only solution when the PiJuice gets into this state seems to be a “hard reset”; pull the battery connector off the board (with external power disconnected). After this the Juice seems to work nicely, once reconnected. 
  
For my application, I’m considering adding a momentary-off “reset” button to the battery positive cable, but would prefer a better solution if possible. 
 
Is there anything on the Juice board (or software) to perform this “hard-reset”? I would also consider foregoing the RTC functionality if that provides more reliable behaviour.

uGiFarukh, 2020-01-13:
Do you have any peripherals or sensors connected to the 3.3V rail of the pijuice? If so, what is the maximum current draw from those? Also, what firmware version and software version are you using at this moment?

GregHibbert, 2020-01-21:
Thanks for replying!

I have an atypical setup - the PiJuice is directly connected to the RPi only by pin 4 & 6; otherwise it is connected via the I2C output on the HyperPixel (I2C bus 3) which includes the 3v3 rail. I've connected these to the PiJuice Zero pins 3 (SDA), 5 (SCL), 9 (GND) & 17 (3v3).

There are no other devices on the I2C bus, just the PiJuice (via the HyperPixel).

I'm not sure if there's any power draw on the PiJuice 3v3 rail as a result of this.

What do you think may be happening here? Is there a reason to think that the 3v3 rail could be the cause of my issues?