PiSupply/PiJuice #426: "Just Power Up"

etherbrat, 2019-12-11:
Suppose the PiJuice has shut down the Pi due to low charge or low battery voltage. I want the Pi to reboot immediately after the A.C. power has been restored - not wait for the charge to replenish. How do I configure the PiJuice to do that?

ChristopherRush, 2019-12-11:
You can set `Wakeup on charge` under `System Task` to 0%. The Pi will then boot when power has been applied.

mchollett, 2019-12-12:
Is there anything that would keep this from working ?

zt4, 2019-12-15:
This doesn't work for me either. It seems like the "Wakeup on charge" needs to pass through the set point before it will turn on, so if the battery is already above the "Wakeup on charge" value, it won't turn on. It works properly when I have "Min charge" set to say 20% and "Wakeup on charge" set to 21%. Then it will power up at 21% after shutting down at 20%.

I shut off the Pi via another program and have "Software Halt Power" enabled to 20s. The Pi shuts down, and 20s later, the battery light shuts off. But I haven't found a setting that will let it turn the Pi back on when re-connecting the power to the PiJuice board when it is above the "Wakeup on charge" value.

My battery level is over 90%, "Wakeup on charge" is set to 15% or 0% and when I plug in the power to the PiJuice board, the charging status lights come on, but it doesn't start the Pi. I also have "No battery turn on" enabled, but that doesn't help either.

`{
  "system_task": {
    "enabled": true,
    "watchdog": {
      "enabled": false,
      "period": 4
    },
    "wakeup_on_charge": {
      "enabled": true,
      "trigger_level": "0"
    },
    "min_charge": {
      "enabled": true,
      "threshold": "10"
    },
    "min_bat_voltage": {
      "enabled": false,
      "threshold": 3.3
    },
    "ext_halt_power_off": {
      "enabled": true,
      "period": "20"
    }
  },
  "system_events": {
    "low_charge": {
      "enabled": true,
      "function": "SYS_FUNC_HALT_POW_OFF"
    },
    "low_battery_voltage": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "no_power": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "watchdog_reset": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "button_power_off": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "forced_power_off": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "forced_sys_power_off": {
      "enabled": false,
      "function": "NO_FUNC"
    }
  }
}
`

tvoverbeek, 2019-12-15:
The wakeup-on-charge flag in the firmware gets reset as soon as it boots the Pi. It has to be commanded set again before the Pi shuts down. This is done by the service (pijuice_sys.py) just before the shutdown command is sent in case the shutdown (SYS_FUNC_HALT_POW_OFF) is triggered by either the 'low_charge", 'low_battery_voltage" or 'no_power' event.
@zt4 in case you shut down the Pi via another program you have to send the pijuice.power.SetWakeUpOnCharge(<trigger level>) command yourself.

zt4, 2019-12-15:
Thanks, @tvoverbeek. I see that you mentioned this here as well. https://github.com/PiSupply/PiJuice/issues/299#issuecomment-476291108

I will try that and let you know if there are any problems.

zt4, 2019-12-15:
@tvoverbeek No issues now! Running the "pijuice.power.SetWakeUpOnCharge(0)" command just before I shut down the Pi, and the Pi starts as soon as the power is re-connected. Thanks!

etherbrat, 2019-12-16:
Thanks for the clarification. I’m getting close to a workable configuration now.

 

Thanks also for your replies to my other queries.

 

Kind regards,

Neil (etherbrat)

neilahiggins@ieee.org

 

@zt4 <https://github.com/zt4>  in case you shut down the Pi via another program you have to send the pijuice.power.SetWakeUpOnCharge() command yourself.

zt4, 2019-12-21:
> @tvoverbeek No issues now! Running the "pijuice.power.SetWakeUpOnCharge(0)" command just before I shut down the Pi, and the Pi starts as soon as the power is re-connected. Thanks!

Unfortunately, this doesn't always work. Twice I've had it not power on. I even modified my program to set the SetWakeUpOnCharge after boot, and twice before shutting down. I'm not checking the return value from that command, which I will try.

Any other thoughts on why this wouldn't work?

Thanks.

etherbrat, 2019-12-23:
Any more on this? I am evaluating the PiJuice for use in remote locations with unreliable power. Unless this function is reliable, I cannot go ahead. 

Neil Higgins (iPhone)
higgins-demack@bigpond.com

zt4, 2019-12-23:
> Any more on this? I am evaluating the PiJuice for use in remote locations with unreliable power. 

I've modified my program to send me an email whenever I set the SetWakeUpOnCharge value with the return value. Since then, I've received the email consistently on startup and shutdown with the return value of {error: NO_ERROR}. And it has turned on each time it was supposed to. I'll keep monitoring it and let you know if I have any further issues.

etherbrat, 2019-12-23:
Many thanks. 

Neil Higgins (iPhone)
higgins-demack@bigpond.com

tvoverbeek, 2019-12-23:
@zt4 @etherbrat Note: There is also a GetWakeUpOnCharge() function. Return value is the threshold. A return value of 127 (0xFF) means disabled.

zt4, 2019-12-24:
Thanks, @tvoverbeek I've added that to my e-mail notification. So far, it shows 0, as it should.

zt4, 2019-12-25:
Unfortunately, this morning it didn't power on the Pi when the power was restored. I got my email notification when it shut down last night showing {error: NO_ERROR} and {data: 0, error: NO_ERROR} for SetWakeUpOnCharge(0) and GetWakeUpOnCharge(). So everything looked like it should power up and it did work the previous night with no changes.

Any ideas why it wouldn't?

Thanks.

zt4, 2019-12-26:
It worked fine for powering on yesterday afternoon. 

Then this morning it did not power on. I checked and the slow blink wasn't even working. I pressed SW1, and it wouldn't turn on. I unplugged power and reconnected it, still wouldn't power on. Plugged power directly into the Pi briefly, and the Pi started, but no lights on the PiJuice. I then brought it indoors, plugged it in, still nothing, then removed the battery and it started. Put the battery back in, and the charging lights are blinking like they should.

The Pi is kept in my car, and the past two nights, the temperature in the car has been around -10C. Not sure if that is the issue maybe.

tvoverbeek, 2019-12-26:
Temperature could be an issue. If temperature drops below tCold (1ºC for BP7X) charging is disabled. At low temperatures battery capacity is lower and the output voltage drops. When the voltage drops too low the MCU shuts down. For wake up to work the MCU has to be running (obviously). So this is probably what happened.

zt4, 2019-12-26:
Thanks. I'll change it to shut down at a higher state of charge, to see if that helps. Right now it is set to shutdown at 25% on the BP7X. I'll try 75% tonight.

mchollett, 2019-12-30:
Any update on this ?  Was it temperature related ?  I'm still having random issues of the Pi not waking back up when the power is restored.  I have not tried setting the wake up on charge flag with my own software.  That is next on the list to try.  Wish there was a way to put this in the firmware so that it always powers back up when power is restored.

zt4, 2019-12-30:
I tested a couple of times with the higher charge. One time the PiJuice powered up and the charging light was on/blinking, but it didn't power on the Pi.

I'm away, so not able to do further testing this week, but it certainly is not reliably turning on. I'll give more updates later.

mchollett, 2020-01-07:
Any updates ?  My setup is still hit or miss..... Mostly miss

zt4, 2020-01-11:
It works fine, as long as the temperature remains basically above 0C. A combination of temperatures below 0C and a low battery level makes it unreliable. Even a relatively high battery level (60%) and -10C makes it unreliable. I shut it down via my own program at 80% now to help ensure it turns back on.

mchollett, 2020-01-31:
Has there been any other discussion on this ?  I'm still having trouble. 

I added to my software:
   pj.power.SetWakeUpOnCharge(0)

My python script sends this every couple of minutes.

The unit was sitting overnight.  I just walked up to it and turned power back on.  RPi did not start up.  Gave it a reasonable amount of time - no startup.  Started the RPi by pushing the button on the PiJuice.  Gave it some time to settle.  Shut off power.  PiJuice properly shutdown RPI.  Waited till all LEDs were off.  Turned power back on and RPi started properly.

It works correctly unless it sits off for an extended period of time.

Is there any way I can modify the firmware so it starts up everytime it sees power ?

This is getting frustrating.  I have one of the older units that works with no problems.

Please help.

Thanks,
Mark

zt4, 2020-01-31:
Have you tried swapping the batteries between the one that works and the one that doesn't work? Based on some similar experiences, if the battery gets too low (may be a weak battery), it won't boot up when power is returned. This is because the boot up on power setting is volatile. So if the battery dies, it doesn't know to turn on.

mchollett, 2020-01-31:
I have not tried that but I have a new PiJuice sitting in front of me that I'll pull the battery out of.

I'll leave it over the weekend and let you know how it goes.

Thanks,
Mark

etherbrat, 2020-02-02:
Please let us know how it goes. I’m wary of complicating our system with a device that might introduce new problems.

 

Thanks,

Neil

mchollett, 2020-02-04:
It looks like replacing the battery solved the problem.  The unit sat for 3 days.  I just went out and turned the power on and the RPi booted as it should.

It is nice to understand what the problem was, but I still have concern about future problems.  If a unit sits idle for an extended period, it may not start back up when power is applied.

If I understand correctly, the startup on power state is a volatile variable held in memory on the PiJuice.  We need a way to insure that the unit will power on the RPi when power is restored regardless of the state of volatile memory on the PiJuice.

I'm guessing the reluctance in changing firmware is that you then have two versions of the firmware that must be maintained.  A "follow the volatile memory" version and a "always power on" version.

Would it be possible to take an input on the PiJuice and use it as an always power up indicator.  It would be acceptable to me to supply +5vDC to the PiJuice via the USB connector and then also supply +5vDC to a input pin to insure that the PiJuice always powers on the RPi when power is restored.

I'm still hesitant to put these in long term projects as the equipment must power on when power is supplied.

Thanks,
Mark

tvoverbeek, 2020-02-04:
Have you experimented with the 'No battery turn on' setting?
When enabled and power returns to the PiJuice (via its USB input) the PiJuice will turn on the 5V and start the Pi, even when no battery is present (i.e Pijuice MCU is off) and probably also when battery is dead.

zt4, 2020-02-05:
I've had mine set with the no battery turn on. I've had issues when it is cold (around -10C).

I've had power on (to the PiJuice USB), but no lights on the board or Pi. I then press the button to power on, and nothing happens. Disconnect and reconnect the input power to the PiJuice and nothing comes on. Then I pop out the battery for a second, and both the PiJuice and Pi power on.

I've also had the PiJuice power on (light on), but not power on the Pi. In that case, I press the PiJuice button, and the Pi turns on.

In cases where I've caught this situation quickly, I've noticed that the battery% is much lower than the level was when I automatically shut down the Pi (80%).

mchollett, 2020-02-05:
> Have you experimented with the 'No battery turn on' setting?
> When enabled and power returns to the PiJuice (via its USB input) the PiJuice will turn on the 5V and start the Pi, even when no battery is present (i.e Pijuice MCU is off) and probably also when battery is dead.

Is this stored in volatile memory ?  If all power is lost, is this setting retained ?

I'm open to trying this.  Can I just change the setting using the CLI and then remove the battery to test ?

Thanks,
Mark

tvoverbeek, 2020-02-05:
The setting is stored in non-volatile memory, so you can remove the battery to test