PiSupply/PiJuice #428: For what functions is the pogo pin REQUIRED?

etherbrat, 2019-12-11:
The latest documentation states that the pogo pin is no longer required under certain circumstances, but it remains unclear for what purposes the pogo pin MUST be installed. Are they the same for the RPi3B+ and the RPi4?

ChristopherRush, 2019-12-11:
Pogo pin was required to do wakeups when the Pi was powered via the Pi's micro USB/USB C port. Since it was introduced we have been able to do this via software. It is documented here - https://github.com/PiSupply/PiJuice/tree/master/Hardware#unpopulated

shawaj, 2020-02-22:
For reference, it now uses the GPIO 3 to do the same thing. See here https://github.com/PiSupply/PiJuice/blob/1f89d55a14879768a644580df3b57a0eaebd0e7c/Software/Source/debian-base/changelog#L39

In the next batches we will likely stop including the pogo pin by default in the packaging as it's not really needed anymore and we don't have positions for 3B+ and 4B on the PCB

closed 2020-02-22 by shawaj