PiSupply/PiJuice #431: Clarification of watchdog reset

etherbrat, 2019-12-17:
1. Is the watchdog reset now achieved in all cases by temporarily shorting GPIO pins 5 & 6? If so:
1a. How is this compatible with the use of GPIO 6 for I2C communications?
1b. Does the installation script ensure that pin 6 is correctly configured for this function (watchdog reset)?
1c. What checks of configuration files can I do to confirm that it has been properly configured?

2. It is also possible that the watchdog reset could be achieved via a power cycle, but only when the Pi is being powered from the HAT's microUSB.
2a. Is this done under any circumstances?

tvoverbeek, 2019-12-17:
@etherbrat pin 6 is ground and pin 5 is GPIO3. When the Pi  is shut down (in low power state) a falling edge on GPIO3 (e.g. shorting pins 5 and 6) will make the GPU firmware to boot the Pi.
Note: this implies the Pi still has 5V power.
This has been the case since the original Pi in 2012.
Since the ARM is not active yet it does not interfere with i2c which is configured during the ARM boot.
So nothing to do to check if things are properly configured.
The watchdog reset uses the same sequence as the other wakeups (interrupt 5V when powered via the PiJuice, otherwise use run pin and finally if no run pin use the GPIO3 method).

etherbrat, 2019-12-18:
Thank you for your clarification, which is very helpful. Could I respectfully request that such explicit detail be added to the product documentation?

shawaj, 2020-10-24:
Added this to readme via https://github.com/PiSupply/PiJuice/commit/709edbace0f5720e3fedf35c1f510e280c29a7ec

commit 709edbac (2020-10-24): Add detail to wake up functionality

As detailed description
Closes #431
closed 2020-10-24 by shawaj