PiSupply/PiJuice #432: PiJuice + Raspberry Pi 4 + LoRa/GPS HAT

gridtechewz, 2019-12-18:
Hi guys,

Setup: I have a Raspberry Pi4 connected with 2 shields (Pi Juice & LoRa/GPS HAT from Dragino) and I have already installed the software packages, in order to see the battery charge level, settings, etc.

I have the following problems: 

1. If I power it up via RP4, the Pi will be turned on automatically, but powering through the PiJuice, it won't start the Pi and only the lights of the PiJuice are blinking.

2. If I pull out the power cable, then the Pi shuts down, despite the battery being almost fully charged. Currently, I have the default settings, do I have to change the settings, in order to run the Pi via battery?

3. I want to send turn on/off notification immediately, when the Pi is powered via power cable, but how can I access the information, whether the GPI-Output is PRESENT/NOT PRESENT, in order to write this information in the LoRa-payload to send it.

Thanks for your attention and help!

shawaj, 2020-01-06:
@gridtechewz did you intentionally close this?

gridtechewz, 2020-01-06:
Hi, since noone answered the question, I thought, I would just close it. Why did you ask?

shawaj, 2020-01-06:
@ChristopherRush can you help @gridtechewz with this?

ChristopherRush, 2020-01-06:
1. You need to press the button SW1 on the PiJuice to boot up the Pi or you
can set Wakeup on Charge with a value of 0%.

2. The battery should take over without any change needed to be made. I
think this issue has been reported elsewhere, will check.

3. You would need to use our API which is documented here -
https://github.com/PiSupply/PiJuice/tree/master/Software You can check the
current status with `pijuice.status.GetStatus()`. When the Pi is turned on
you could just run a boot up script with cron. When the power cable has
been removed use the `NO POWER` event to run a user script.

gridtechewz, 2020-01-07:
> 1. You need to press the button SW1 on the PiJuice to boot up the Pi or you can set Wakeup on Charge with a value of 0%. 2. The battery should take over without any change needed to be made. I think this issue has been reported elsewhere, will check. 3. You would need to use our API which is documented here - https://github.com/PiSupply/PiJuice/tree/master/Software You can check the current status with `pijuice.status.GetStatus()`. When the Pi is turned on you could just run a boot up script with cron. When the power cable has been removed use the `NO POWER` event to run a user script.
> […](#)
> -- Kind Regards *Chris Rush*

Thank you for your answer!

gridtechewz, 2020-01-09:
> 1. You need to press the button SW1 on the PiJuice to boot up the Pi or you can set Wakeup on Charge with a value of 0%. 2. The battery should take over without any change needed to be made. I think this issue has been reported elsewhere, will check. 3. You would need to use our API which is documented here - https://github.com/PiSupply/PiJuice/tree/master/Software You can check the current status with `pijuice.status.GetStatus()`. When the Pi is turned on you could just run a boot up script with cron. When the power cable has been removed use the `NO POWER` event to run a user script.
> […](#)
> -- Kind Regards *Chris Rush*

Hi Chris, 
1. It worked perfectly! Unfortunately, my USB Micro power input was WEAK. The reason is probably that I couldn't fully plug in my USB Micro cable into the Pijuice. So, I decided to use the GPIO power input. 
2. I thought maybe it is the power consumption, why it didn't work. So, I removed the GPS/LoRa HAT, suddenly my battery fell down and when I put it back on the Pijuice, the light appeared and I could run RP4 with battery only and then finally also with the GPS/LoRa HAT. So, the issue was not the power consumption, but the battery was not close enough to the contact point to provide electricity, despite plugged into the pijuice. Unfortunately, another issue has risen. I can now start RP4 via battery only or start it by plugging the GPIO power cable and it charges the battery as well. If I remove the power cable, the battery will automatically run the RP4 without problem, **but as soon as I plug in the GPIO power cable in, it won't recognize it and I can't charge the battery again.** 

Do you know, what might be the problem?

3. Still working on it!

tvoverbeek, 2020-01-09:
What do you mean with GPIO power cable?
Power connected to J4? or directly to the GPIO pins?
If directly to the 5V GPIO pins it is equivalent with power connected to the RPi.

gridtechewz, 2020-01-09:
> What do you mean with GPIO power cable?
> Power connected to J4? or directly to the GPIO pins?
> If directly to the 5V GPIO pins it is equivalent with power connected to the RPi.

I refer to the 5V GPIO Pins, which basically is the power connected to the RPi like you said.

tvoverbeek, 2020-01-09:
@gridtechewz Check in the settings (In GUI Configure HAT -> General) that GPIO input is enabled.

gridtechewz, 2020-01-13:
> @gridtechewz Check in the settings (In GUI Configure HAT -> General) that GPIO input is enabled.

It is already enabled. Do you have other ideas?

tvoverbeek, 2020-01-13:
Assuming the Pi remains running when on battery, when youplug in the GPIO power cable again what does the GUI/CLI say about power input? I am expecting GPIO power input present. Note when power input is reported as WEAK, it will not charge the battery.

gridtechewz, 2020-01-14:
Well, here I have my configuration settings!
[image]

When I start the RPI4 with the GPIO Power cable inside, then it looks like this:
[image]

When I unplug the power cable and the RPI4 runs only through battery, it looks like this:
[image]

When I put the power cable back, it looks like this:
[image]

gridtechewz, 2020-01-14:
Here you see my settings:
[image]

Then when I start the RPI4 with the GPIO power cable, it looks like this:
[image]

When I unplug the power cable, run it only via battery, it looks like this:
[image]

When I put the power cable back, it looks like this:
[image]

So, it does not even detect the cable anymore for some weird reason. Any ideas? Why?

tvoverbeek, 2020-01-14:
Negative IO current means current flows from the Pi to the PiJuice (charging)
Positive IO current flows from the PiJuice to the Pi.
When on battery the PiJuice supplied 2.4A to the Pi (which is probably the max it can supply).
On your last picture the current to the Pi is 2.9A.
Any heavy current peripherals connected (e.g. usb powered hard disks)?
It seems your GPIO power supply has no reserve any more to charge the battery.

gridtechewz, 2020-01-17:
> Negative IO current means current flows from the Pi to the PiJuice (charging)
> Positive IO current flows from the PiJuice to the Pi.
> When on battery the PiJuice supplied 2.4A to the Pi (which is [...]

Well the only thing I have connected to the RPI4 + Pijuice is the following: Mouse, keyboard, screen. This setup didn't change in these 3 cases (power cable plugged in, battery only, battery + power cable). I took the screenshots in less than a minute for these 3 cases. So, what do you think of this?

tvoverbeek, 2020-01-17:
@gridtechewz Probably the current measurement gets confused by external power coming via the GPIO 5V pins.
Since you suspected a bad USB power connection to the PiJuice (which caused you to use the GPIO pins) have you considered using the power input via J4 (https://github.com/PiSupply/PiJuice/tree/master/Hardware#top-of-the-board) ?
You need to solder a 2-pin header first. This is equivalent to using the power input via USB.

gridtechewz, 2020-01-17:
> @gridtechewz Probably the current measurement gets confused by external power coming via the GPIO 5V pins.
> Since you suspected a bad USB power connection to the PiJuice (which caused you to use the [...]

Before I solder anything, since I don't have much experience in it.

What do you think about buying a new micro USB cable? Since the one I used for the PiJuice only provides this much 5V == **1A** and I saw in the hardware section, it requires **2.5A**. I think that might be reason, why it was WEAK. 
Do you think if the cable works via PiJuice instead of RPI4, I won't have the problem from before, we tried to solve?

gridtechewz, 2020-01-17:
> @gridtechewz Probably the current measurement gets confused by external power coming via the GPIO 5V pins.
> Since you suspected a bad USB power connection to the PiJuice (which caused you to use the [...]

Hey, I changed adapter to the cable for Pijuice, which provides 2.1A instead of 1A and now it works!
I can run with with cable or plug it off and run it by battery. Finally, when I put the cable back, it charges!!

closed 2020-01-17 by gridtechewz