PiSupply/PiJuice #433: Ubuntu /arm64 compatibility

promethe42, 2019-12-21:
Hi,

could you please provide some pointers/documentation on how to use the PiJuice software on Ubuntu/arm64.

All the software prebuilt/stored in this repo seams to be targeting `armhf`:

```bash
$ file pijuice_cli
pijuice_cli: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=6e5b1ad06aa0ad2a337fb3b585e28beac17df547, not stripped
```

Thank you for your help,

tvoverbeek, 2019-12-22:
@promethe42 The only problem are the setuid wrappers for pijuice_cli.py and pijuice_gui.py.
You have to recompile these for 64 bit.
See https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Setuid-Wrapper for the source and the build script.

promethe42, 2019-12-23:
I ran `build-setuid-progs.sh` (which produced no output). The following files were built:

* `pijuice_cli`
* `pijuice_gui`

```bash
$ ls
build-setuid-progs.sh  pijuice_cli  pijuice_gui  setuid-prog.c
```
but `pijuice_cli` won't run:

```bash
$ ./pijuice_cli                                                                                                                                                                                            stat: No such file or directory
```

"No such file or directory" yet the binary file does exist and has +x.

```
$ file pijuice_cli                                                                                                                                                                                          
pijuice_cli: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=b1a767832e5dc248368db7b6be27473c6f2ca6ab, for GNU/Linux 3.7.0, not stripped
```

promethe42, 2019-12-23:
Found the problem: the generated binary tries to run `/usr/bin/pijuice_cli.py` which was not installed.
Fixed it with `sudo /usr/bin/pijuice_cli.py`.

Now `pijuice_cli` says:

```bash
$ ./pijuice_cli 
./pijuice_cli: /usr/bin/pijuice_cli.py has the wrong owner
The script should be owned by ubuntu.
```

promethe42, 2019-12-23:
The following works:

```sudo -u pijuice ./pijuice_cli```

but it's kinda weird that I can't run it directly.

tvoverbeek, 2019-12-24:
@promethe42
The idea is that the pijuice service (pijuice_sys.py) and the gui an cli run as a non-privileged
separate user (user pijuice). This to prevent user scripts to run as root.
Originally the service (which executes the user scripts) was running as root.
The pijuice_cli and pijuice_gui wrappers should be suid pijuice.
Look at the post-install scripts for Raspbian: https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/postinst
and 
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-gui/postinst

shawaj, 2020-02-22:
@promethe42 did you get this working?

MarkusLange, 2020-03-29:
This little script worked for me,

on an fresh ubuntu server 64bit.

```
#!/bin/bash

sudo apt update
sudo apt install -y python3-smbus python3-pip python3-stdeb dh-systemd debhelper python3-stdeb fakeroot python3-all
pip3 install stem stdeb3

sudo git clone https://github.com/PiSupply/PiJuice.git

cd $HOME/PiJuice/Software/Source/Setuid-Wrapper
sudo ./build-setuid-progs.sh
sudo cp p* $HOME/PiJuice/Software/Source/bin

cd $HOME/PiJuice/Software/Source/
sudo ./pckg-pijuice.sh

sudo cp $HOME/PiJuice/Software/Source/deb_dist_base/pi*.deb $HOME
sudo cp $HOME/PiJuice/Software/Source/deb_dist_gui/pi*.deb $HOME

sudo rm -rf PiJuice

sudo apt install python3-urwid lua5.3
```

JohnMcW, 2020-03-31:
> This little script worked for me,
> 
> on an fresh ubuntu server 64bit.

Worked for me to, but I've no idea what to do next.  I just want to be able to run pijuice_cli

Thanks!

tvoverbeek, 2020-04-01:
@JohnMcW Install the pijuice-base and pijuice-gui deb packages which are in your home directory after running the script.

MarkusLange, 2020-04-01:
> > This little script worked for me,
> > on an fresh ubuntu server 64bit.
> 
> Worked for me to, but I've no idea what to do next. I just want to be able to run pijuice_cli
> 
> Thanks!

```
dpkg -i pijuice-base_1.6_all.deb
dpkg -i pijuice-gui_1.6_all.deb
```

And add
`dtoverlay=i2c-rtc,ds1307`
in the /boot/config.txt

and sync the RTC with the systemclock and check it
```
sudo hwclock -w
sudo hwclock -r
```

JohnMcW, 2020-04-04:
Hi again,

I thought I’d check that the PiJuice works in Raspbian – it does and I also updated the firmware to 1.4

But I need to use Ubuntu 18.04.4 64bit for Arm – so starting with a clean updated install, I ran the script above OK, then:

> dpkg -i pijuice-base_1.6_all.deb

This installed fine!

> dpkg -i pijuice-gui_1.6_all.deb (only if you need the GUI for the desktop)

Don't need/want this.

> And add
> dtoverlay=i2c-rtc,ds1307
> in the /boot/config.txt

Not done as firmware is 1.4

But when I run:
`pijuice_cli`

The GUI starts but errors with:

`"No PiJuice found or Failed to use I2C bus.  Check if I2C is enabled"`

But:

`sudo i2cdetect -y 1`

Displays:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
Any ideas?

tvoverbeek, 2020-04-04:
I do not know how i2c access is arranged on Ubuntu
On raspbian /dev/i2c-x has rw access for group i2c.
If a user belongs to the i2c group the user does not need to be root to access the i2c bus.
Since you use sudo to run i2cdetect it looks like normal users have no i2c access.
Have you tried with `sudo pijuice-cli`?

JohnMcW, 2020-04-04:
> Have you tried with `sudo pijuice-cli`?

No, but I have now and it works! Thank you!

MarkusLange, 2020-04-05:
> > And add
> > dtoverlay=i2c-rtc,ds1307
> > in the /boot/config.txt
> 
> Not done as firmware is 1.4

Firmware v1.4 is about dtoverlay=i2c-bcm2708

JohnMcW, 2020-04-08:
Hi again.  I'm trying to run a Python script - just using SW2 to test, but I'm getting an error (and I suspect it has something to do with permissions, so I thought I'd post here).  This is the code:
```
#!/usr/bin/python3
import sentry_sdk
sentry_sdk.init("https://xxxxxxx@oyyyyy.ingest.sentry.io/zzzzz")
sentry_sdk.capture_exception(Exception("Clicked button SW2 using Python3 "))
```

Note that I've obfuscated some parameters in the init call above.  
This is using sentry.io - an error collection service
And this is the error:
```
Apr 08 20:25:54 ubuntu sudo[4210]:  pijuice : TTY=unknown ; PWD=/var/lib/pijuice ; USER=ubuntu ; COMMAND=/home/ubuntu/sentry/pjsentry.py SINGLE_PRESS SW2
Apr 08 20:25:54 ubuntu sudo[4210]: pam_unix(sudo:session): session opened for user ubuntu by (uid=0)
Apr 08 20:25:55 ubuntu pijuice_sys.py[1358]: Traceback (most recent call last):
Apr 08 20:25:55 ubuntu pijuice_sys.py[1358]:   File "/home/ubuntu/sentry/pjsentry.py", line 2, in <module>
Apr 08 20:25:55 ubuntu pijuice_sys.py[1358]:     import sentry_sdk
Apr 08 20:25:55 ubuntu pijuice_sys.py[1358]: ModuleNotFoundError: No module named 'sentry_sdk'
Apr 08 20:25:55 ubuntu sudo[4210]: pam_unix(sudo:session): session closed for user ubuntu
```

The script is executable and works fine when run manually.

Any ideas?

tvoverbeek, 2020-04-09:
@JohnMcW The error log shows that pijuice_sys.py (= the pijuice service) has started your user function, but the `import sentry_sdk` statement fails.
Remember the pijuice service is started as user pijuice by systemd.
My guess is that the entry_sdk module is not in the default python path.
Where is the module located? When you run it manually do you e.g. use environment variables to add to the python path?

JohnMcW, 2020-04-10:
Hi again @tvoverbeek  - a colleague helped me resolve this - all I need to do was install the sentry software as root.  PiJuice is now successfully running Python scripts!  However, I have a few more questions (hope it's OK to ask here).

1. Does the PiJuice need to be powered directly for 'Min Charge'/'Wakeup on charge' to function correctly?  It appears that it does:

GPIO POWERED

- Remove power to Pi
- “No power” is triggered
- “Low charge” = SYS_FUNC_HALT_POW_OFF
- This is triggered when “Min Charge” threshold is reached
- Pi fully powers down
- Reconnect power to Pi
- Pi immediately powers up (undesired)
- Pi immediately halts (but doesn’t fully power down and so get stuck here)
- After a manual reboot, neither “Forced power off” or “Forced sys power off” are triggered

DIRECT POWERED

- Remove power to Pi
- “No power” is triggered
- “Low charge” = SYS_FUNC_HALT_POW_OFF
- This is triggered when “Min Charge” threshold is reached
- Pi fully powers down
- Reconnect power to Pi
- Nothing happens immediately (desired)
- Instead “Wakeup on charge” kicks in after the wakeup charge threshold is reached (desired)
- However, neither “Forced power off” or “Forced sys power off” are triggered

2. Why aren't “Forced power off” or “Forced sys power off” being triggered?  Should they be and what's the difference between them.

3. Is direct power to PiJuice required for the watchdog functionality?  [This post of yours](https://github.com/PiSupply/PiJuice/issues/209) suggests that it is.  Also, is there an easier way to test?

Thank you

tvoverbeek, 2020-04-10:
> Does the PiJuice need to be powered directly for 'Min Charge'/'Wakeup on charge' to function correctly? It appears that it does:

Yes, for obvious reasons. To control the 5V supply to the Pi the power has to come from the PiJuice.

>Why aren't “Forced power off” or “Forced sys power off” being triggered? Should they be and what's the difference between them.

They are different from low charge and no power.
forced power off occurs when the battery cannot supply the current draw needed by the Pi.
Without anything programmed just let the Pi run on battery until it shuts off. This will be a 'forced power off'
Forced sys power off is the same thing for the extra load on the VSYS output on the P3 header and J3 connector.

>.Is direct power to PiJuice required for the watchdog functionality? This post of yours suggests that it is. Also, is there an easier way to test?

As explained in that post the watchdog gets reset by each I2C transaction to the PiJuice, provided the watchdog is enabled. To test it you need to make sure that no I2C transactions are made to the PiJuice and the watchdog is enabled. My post gave a way to make this possible in a controlled way. If for some reason the Pi goes haywire (endless loop on all cores) the PiJuice watchdog would kick in (again only if enabled). For the PiJuice watchdog reboot to work powering the PiJuice is the safest way.

Note, the Pi has a built-in watchdog. See e.g. https://www.raspberrypi.org/forums/viewtopic.php?t=210974#p1458574.

smallhacks, 2020-07-22:
Hello,

I managed to install pijujicebase on my Ubuntu 64-bit by following the script suggested by @MarkusLange.
Everything seems to work except pijuiceboot  
`-bash: /usr/bin/pijuiceboot: No such file or directory`

It seems that the executable is still a 32-bit version:
`/usr/bin/pijuiceboot: ELF 32-bit LSB executable, ARM`

 Do you know how to compile it for 64-bit?

Thank you!

tvoverbeek, 2020-07-22:
The source is here: https://github.com/PiSupply/PiJuice/blob/master/Firmware/pijuiceboot.c
Just compile it on your 64-bit system. Ignore the comments in the header, libwebsockets is not used.

smallhacks, 2020-07-25:
> The source is here: https://github.com/PiSupply/PiJuice/blob/master/Firmware/pijuiceboot.c
> Just compile it on your 64-bit system. Ignore the comments in the header, libwebsockets is not used.

Thank you very much, it worked !

MarkusLange, 2020-08-02:
> 
> 
> The source is here: https://github.com/PiSupply/PiJuice/blob/master/Firmware/pijuiceboot.c
> Just compile it on your 64-bit system. Ignore the comments in the header, libwebsockets is not used.

@tvoverbeek 
May you can add
gcc pijuiceboot.c -Wall -o pijuiceboot
to the build-setuid-progs.sh so a new pijuiceboot would build anyway
and add my script nicely/some revision to

PiJuice/Software/OS Support/

as universal for all/other Pi OS

tvoverbeek, 2020-08-02:
@MarkusLange Thanks for the suggestion. Updated the 64-bit instructions with a slightly modified version of your script.
Also used apt to install the local version, which will also install the dependencies.
Closing the issue now.

closed 2020-08-02 by tvoverbeek

MarkusLange, 2020-08-02:
Thanks,

from https://github.com/PiSupply/PiJuice/issues/498#issuecomment-624042639 sudo apt install dh-python is missing in 

Software/OS Support/ubuntu_server_64.MD

for Ubuntu 20.04

tvoverbeek, 2020-08-03:
@MarkusLange Added  now.

goodelyfe, 2021-06-09:
on Ubuntu 21.04 (Hirsute) the dh-systemd package has been completely removed, so it fails to compile there because of a missing dependency

MarkusLange, 2021-06-09:
@goodelyfe if @tvoverbeek removes the dependencies to dh-systemd there is a way, or you can try [this](https://stackoverflow.com/questions/54206929/apt-get-install-of-dh-systemd-fails-due-to-debhelper-but-debhelper-is-already-a) and install an backport from dh-systemd from groovy (20.10) the last one who has it.

goodelyfe, 2021-06-09:
> @goodelyfe if @tvoverbeek removes the dependencies to dh-systemd there is a way, or you can try [...]

sudo apt-get install dh-systemd -t groovy ?

or would it be

sudo apt-get install dh-systemd -t groovy-backports

either way i get:

Reading package lists... Done
E: The value 'groovy-backports' is invalid for APT::Default-Release as such a release is not available in the sources

Reading package lists... Done
E: The value 'groovy' is invalid for APT::Default-Release as such a release is not available in the sources

CybrHare, 2021-10-06:
Has anyone happed to figure out how to get this to work on Ubuntu 21.04?

tvoverbeek, 2021-10-09:
dh-systemd has been integrated into debhelper.
Already in buster dh-systemd ws a transitional package which depended on debhelper.
In bullseye this transitional package was removed.
See e.g. https://github.com/matrix-org/synapse/issues/9073
I assume Ubuntu 21.xx is derived from Debian bullseye.

Anyay we will get the same problem when RaspberryPi OS moves to bullseye (with the same solution).