PiSupply/PiJuice #434: Schedule power up when no power from micro usb

alanroche2015, 2019-12-22:
Hi 

Is it possible to schedule the pi power on when no input is supplied ? I have a raspberry pi in the car and want to schedule startup in the middle of the night for updates but the car not be supplying power to the rpi. 

Thanks 

Al

tvoverbeek, 2019-12-22:
Provided the PiJuice Battery has sufficient charge for your updates it can be done by using the alarm function. See https://github.com/PiSupply/PiJuice/tree/master/Software#wakeup-alarm-menu.
You have to select the 'wakeup enabled' option.

Note the RTC time is in UTC, not your local time. To set your alarm the alarm time also has to be in UTC.

If your Pi is connected to the internet there is another complication.
With the default settings the OS will also access the RTC as a ds1339. The timesyncd service will synchronise the internet time to the RTC  and **clear the alarm flag**, and your wakeup will not work.
To avoid this set the I2C address of the HAT EEPROM to 0x52 instead of 0x50. Then the device-tree fragment for the ds1339 RTC will not be loaded. If you want the time synchronisation at boot (when connected to the internet) add the following line to /boot/config.txt: `dtoverlay=i2c-rtc,ds1307`. The ds1307 has the same commands as the ds1339, but **no alarm function**, hence no interference with the wakeup flag.

When your Pi is not connected to the internet you will have to set the RTC time yourself.
When the Pi has the correct time use `sudo hwclock --systohc` See man hwclock.

shawaj, 2020-10-24:
Closing this issue as no activity in nearly a year. @alanroche2015 feel free to reply if you disagree

closed 2020-10-24 by shawaj