PiSupply/PiJuice #435: generate an Interrupt for Raspberry Pi via GPIO

ThScho, 2019-12-28:
I want to generate an interrupt for the Raspberry Py triggered from an external sensor (optocoupler switch). I therefore have these questions:
1. Can I generate an interrupt using the IO-Pins at P3 (through the STM32-F0)?
2. If not: Are there unused Raspi GPIO pins that I could use (soldering a wire directly on the 40 pin connector which unfortunately is completely occupied by the pijuice)?
I'm using PiJuice Zero.
Thanks, Thomas.

tvoverbeek, 2019-12-28:
@ThScho 
1) No. There are no provisions in the firmware for this.
2) Much easier to use the RPi GPIOs directly. The PiJuice Zero only uses the power pins and the I2C clock and data (GPIO2 and 3, pins 3 and 5)
Use an extension header with long pins  (e.g https://www.adafruit.com/product/2223) for a standard Pi or a 40 pin header with long pins on a PiZero. You can then press down the PiJuice Zero so the pins stick out above the PiJuice Zero and are available to connect to your optocoupler switch.

ThScho, 2019-12-28:
Perfect, thanks! I'll either use the requested Adafruit header or I even solder a wire on the bottom of the Pi. It is important for me to know that there are (lots of) unused GPIOs.

closed 2019-12-28 by ThScho