PiSupply/PiJuice #436: PiJuice Zero GPIO

promethe42, 2020-01-01:
Hi,

based on this picture:

https://i1.wp.com/learn.pi-supply.com/wp-content/uploads/2019/02/20190221_140100.jpg?w=1000&ssl=1

GPIO pins are masked/trapped in the PiJuice Zero hat.

How is one expected to keep using the GPIO pins when using the PiJuice Zero hat?

Regards,

ChristopherRush, 2020-01-01:
Use a extending GPIO header.

promethe42, 2020-01-01:
Here are the solutions I found:

1. Longer GPIO pins

https://www.amazon.com/uxcell-Straight-Connector-Arduino-Prototype/dp/B07DJYYPF3/ref=pd_sbs_147_1/136-3780151-6181925?_encoding=UTF8&pd_rd_i=B07DJVHXY8&pd_rd_r=c8251c16-6a69-4e6d-84aa-4d37f158dc89&pd_rd_w=Wwu9z&pd_rd_wg=IsF9F&pf_rd_p=7c0dad87-8a25-4c4f-9349-026039ea6cb3&pf_rd_r=AERXJ7DGJ3VHYFPV46GM&refRID=AERXJ7DGJ3VHYFPV46GM&th=1

2. 1 to 2 GPIO extension

https://www.amazon.com/Connectors-Raspberry-40-pin-Expansion-RAS-GP02/dp/B07MCW4KCM/ref=pd_sbs_147_4/136-3780151-6181925?_encoding=UTF8&pd_rd_i=B07MCW4KCM&pd_rd_r=c8251c16-6a69-4e6d-84aa-4d37f158dc89&pd_rd_w=Wwu9z&pd_rd_wg=IsF9F&pf_rd_p=7c0dad87-8a25-4c4f-9349-026039ea6cb3&pf_rd_r=AERXJ7DGJ3VHYFPV46GM&psc=1&refRID=AERXJ7DGJ3VHYFPV46GM

By "a extending GPIO header", do you mean solution 1 or 2 ?

Regards,

ChristopherRush, 2020-01-01:
Solution 1, use a stacking gpio header.

tvoverbeek, 2020-01-02:
Better than your reference for 1) (unless you use a Pi Zero without header) is e.g. this GPIO STacking Header: https://www.adafruit.com/product/2223
You can press down the PiJuice Zero Hat so the extra long pins stick out above the PiJuice Zero.
Same goes for your amazon reference when using a Pi Zero without header.

promethe42, 2020-01-03:
Yeah I bought the Amazon ref as a replacement for the default Pi Zero pins. It requires extra soldering but should take less height overall.

I'll post the result here.

shawaj, 2020-02-22:
@promethe42 how did you get on?

brettbeeson, 2020-04-22:
I'd also like to access my PiZero's GPIO pins when using a PiJuiceZero. Can you confirm this the  recommended approach:

**PiJuice with existing headers:**  use a female to extra-long male extension header on the top, like [this simple one]( https://www.adafruit.com/product/2223) or [this complex one](https://www.amazon.com/Connectors-Raspberry-40-pin-Expansion-RAS-GP02/dp/B07MCW4KCM/ref=pd_sbs_147_4/136-3780151-6181925?_encoding=UTF8&pd_rd_i=B07MCW4KCM&pd_rd_r=c8251c16-6a69-4e6d-84aa-4d37f158dc89&pd_rd_w=Wwu9z&pd_rd_wg=IsF9F&pf_rd_p=7c0dad87-8a25-4c4f-9349-026039ea6cb3&pf_rd_r=AERXJ7DGJ3VHYFPV46GM&psc=1&refRID=AERXJ7DGJ3VHYFPV46GM). Just push it on to existing PiZero pins.

**PiJuice without existing headers** (i.e. just holes): use a new extra long male [like this](https://www.amazon.com/uxcell-Straight-Connector-Arduino-Prototype/dp/B07DJYYPF3/ref=pd_sbs_147_1/136-3780151-6181925?_encoding=UTF8&pd_rd_i=B07DJVHXY8&pd_rd_r=c8251c16-6a69-4e6d-84aa-4d37f158dc89&pd_rd_w=Wwu9z&pd_rd_wg=IsF9F&pf_rd_p=7c0dad87-8a25-4c4f-9349-026039ea6cb3&pf_rd_r=AERXJ7DGJ3VHYFPV46GM&refRID=AERXJ7DGJ3VHYFPV46GM&th=1). You gotta solder this.

tvoverbeek, 2020-04-23:
@brettbeeson 
**PiJuice with existing headers:** Yes, those works.
**PiJuice without existing headers:** If you use an extra long header then you have the risk you press down the PiJuiceZero too low so it touches the Pi. Recommend to use a normal header and put an extension header on top as above.

shawaj, 2020-10-24:
Just be careful with the Adafruit ones because they have pins that are not square (normally 0.4 x 0.64) and so they don't always get a great connection.

Toby electronics have some good quality ones https://www.toby.co.uk/board-to-board-pcb-connectors/254mm-sockets/ref-raspberry-pi-rpi-hat-specification-connector-extended-tail-sockets/REF-182683-02/

shawaj, 2020-10-24:
That last link seemed to be a duff link. Here is the correct one https://www.toby.co.uk/board-to-board-pcb-connectors/254mm-sockets/ref-raspberry-pi-rpi-hat-specification-connector-extended-tail-sockets/

closed 2020-10-24 by shawaj