PiSupply/PiJuice #437: Pijuice send email when power is removed

mikcatta, 2020-01-02:
Rather than do just a shutdown, I would like to send email, wait for a few seconds for it to be sent, and then shutdown.

I thought the PiJuice WhatsApp example, might of given some clue as to how to do this, but it doesn't seem to show how to integrate the script into the power loss process.

tvoverbeek, 2020-01-02:
Link the 'No Power' event on the System Events Tab to e.g. USER_FUNC2 (one of the 15 available).
On the User Scripts tab define USER_FUNC2 with the name of your script which should send the email, wait a few seconds and then shutdown the Pi.
In the WhatsApp example they did not use the pijuice service, but detected the no power with low level calls.

mikcatta, 2020-01-03:
Thanks, it was the shutdown script I was particularly struggling with, but it seems it as easy as :  - 

send email and then....
> from subprocess import call
> call("sudo shutdown -h now", shell=True)

closed 2020-01-03 by mikcatta