PiSupply/PiJuice #438: Using script to sendemail and shutdown doesn't actually shutdown

mikcatta, 2020-01-03:
Been playing with sending an email, which is triggered by the the system event "No power" being linked to "USER_FUNC2".

I get the email

mikcatta, 2020-01-03:
Think I've found an answer in this:

#!/usr/bin/env python

import os
from pijuice import PiJuice

pj = PiJuice(1, 0x14)

# Remove Pi power after 30 seconds
pj.power.SetPowerOff(30)

# Shutdown
os.system("sudo halt")

mikcatta, 2020-01-03:
Well that helps with the shutdown, but if I re-apply power the Pi doesn't startup automatically! 

Changed the script to do the following and this seems to work, but more testing required;-```

pj.power.SetWakeUpOnCharge(80)
pj.power.SetPowerOff(30)
```

tvoverbeek, 2020-01-03:
@mikcatta Yes, if you shut down from your own script you have to set the WakeUpOnCharge yourself.
Note that the pijuice service when executing the SYS_FUNC_HALT_POW_OFF does exactly the same as your script including setting the WakeUpOnCharge to the level you have specified in the gui/cli (and wakeup on charge is enabled in the gui/cli)
So you could connect the no power event to SYS_FUNC_HALT_POW_OFF (No user script needed)

mikcatta, 2020-01-06:
Is there a way to get the WakeupOnCharge level specified in the gui/cli?

tvoverbeek, 2020-01-06:
@mikcatta In Python:
```
from pijuice import PiJuice
pj=PiJuice(1,0x14)
pj.power.GetWakeUpOnCharge()
```
The data field is the percentage level.
A value of 127 means disabled.

mikcatta, 2020-01-07:
My script returns {'data': 127, 'error': 'NO_ERROR'} from the pj.power.GetWakeUpOnCharge(), but in the configuration, I have set the wakeup on charge level in the GUI to 80%. I am doing something wrong?

tvoverbeek, 2020-01-07:
127 (0x7F) means WakeUpOnCharge disabled.
The setting in the gui (both enable status and threshold) are stored in /var/lib/pijuice/pijuice_config.JSON.
The system task uses this to set the WakeUpOnCharge just before shutdown when it executes  SYS_FUNC_HALT or SYS_FUNC_HALT_POW_OFF. In your case you still have to set it yourself just before sending the shutdown command, i.e. `pj.power.SetWakeUpOnCharge(80)`.
Let me know if you want to know how to pick up the value and enable status for wake up on charge from the configuration file.

mikcatta, 2020-01-07:
Yes, I understand the the value I get returned from the WakeupOnCharge command is saying it is disabled, but why is not showing as disabled in the GUI (as per the screenshot)?

An extract from the config file gives:- 

"wakeup_on_charge": {
      "enabled": true,
      "trigger_level": "80"
    },

I am missing something here?

tvoverbeek, 2020-01-07:
@mikcatta As I wrote before, the setting in the gui (or cli) only applies when using SYS_FUNC_HALT or SYS_FUNC_HALT_POW_OFF,  e.g. when coupling the low charge event to SYS_FUNC_HALT_POW_OFF.

mikcatta, 2020-01-08:
So in summary:
Using the SYS_FUNC_HALT or SYS_FUNC_HALT_POW_OFF, the PiJuice will be operated correctly, but doesn't allow an email to be sent.
If I use a USER_SCRIPT to perform the email & shutdown, I get a permissions message "Permission denied: '/var/lib/pijuice/pijuice_config.JSON'" whilst trying to read the file, whilst executing the following line of code "_with open(piJuiceConfigDataPath, 'r') as outputConfig:_"

Is there a way to achieve normal operation of the PiJuice & set the wake-up on charge value according to the value specified in the config file from my Python script?

tvoverbeek, 2020-01-08:
pijuice_config.JSON is only accessible for users pijuice and root.
If you want to pick up the wake up on charge threshold from the config file use something like this example:
```
#!/usr/bin/python3

import subprocess
import json
from pijuice import PiJuice

pj = PiJuice(1, 0x14)

config = subprocess.getoutput("sudo -u pijuice cat /var/lib/pijuice/pijuice_config.JSON")
configData = json.loads(config)

if (configData.get('system_task', {}).get('wakeup_on_charge', {}).get('enabled', False)
    and 'trigger_level' in configData['system_task']['wakeup_on_charge']):
   tl = float(configData['system_task']['wakeup_on_charge']['trigger_level'])
   print("Set wake up on charge trigger level to " + str(tl))
   pj.power.SetWakeUpOnCharge(tl)
```

mikcatta, 2020-01-08:
Thank-you, that has worked a treat :)

closed 2020-01-08 by mikcatta