PiSupply/PiJuice #439: Set wakeup alarm for long durations

Jrob0318, 2020-01-06:
Currently I have the PiJuice working by waking up once every 5 minutes.  The Pi then runs a python code with crontab that controls a camera to take a picture which then uploads to dropbox.  The picture is then removed from the storage memory of pi, the wakeup alarm is reset to turn on, and the Pi halts functions so PiJuice can cut power to the board.  The PiJuice is connected to a small solar panel that will keep the battery charged when work is not being done and the data connection is achieved by a sixfab base HAT with a QUECTEL EC25-V cellular modem with a Verizon data plan.  The GUI is kind of confusing me for setting the wake time for long durations.  If I want the PiJuice to wake up 1 time per week or 1 time per month what configuration would the setup look like in the GUI?

tvoverbeek, 2020-01-06:
@Jrob0318 
For once a week Select Weekday and specify a day number (1..7) and Hour, Minute (No Minutes period!), Second as required.
For once a month Select Day and specify the day number (1..28) if you want it every month.

Jrob0318, 2020-01-07:
Thank you very much! This worked properly in the test that I ran yesterday.

closed 2020-01-07 by Jrob0318

quitmeyer, 2024-03-12:
> For once a week Select Weekday and specify a day number (1..7) and Hour, Minute (No Minutes period!), Second as required.
> For once a month Select Day and specify the day number (1..28) if you want it every month.

I'm having a tricky time with the alarm setting, there seems to be something odd with the 'day' key

for instance, this works fine 
`pj.rtcAlarm.SetAlarm({'second': 0, 'minute': 0, 'hour': '0;4', 'weekday': '1;4'})`

but this doesn't work
`pj.rtcAlarm.SetAlarm({'second': 0, 'minute': 0, 'hour': '0;4', 'day': '1;4'})`

however, if only one number is in 'day' it will work
 pj.rtcAlarm.SetAlarm({'day': '3'})

and by "work" i mean when i open the GUI and look at wakeup alarm, the values have changed. Am i doing something wrong here?

rpi4 bookworm, fresh install march 01