PiSupply/PiJuice #440: Errors when installing on headless Pi 3b+

marthazimet, 2020-01-06:
I purchased PiJuice HAT to evaluate for my production solar devices in the field. Obviously, they are all headless. 

Following the instructions for install without GUI, when I invoke pijuice_cli
 there is an error. Screen dump attached. 

[image]

tvoverbeek, 2020-01-06:
@marthazimet  Looks like you are runing as root.
You have to run pijuice_cli and pijuice_gui as a non privileged user.

marthazimet, 2020-01-06:
Thanks... I was following the online docs and didn't see that mentioned. 

I have solar devices deployed in the field and I am looking for a replacement for WittyPi boards which have been failing after about six months. 

This is my device which is deployed in remote sites in the US, Canada, and recently in NZ: https://manray.com/electronic-game-bird-recall

PiJuice looks promising.

closed 2020-01-06 by marthazimet