PiSupply/PiJuice #441: Pijuice button push not detected

thomerc1, 2020-01-07:
Using a several Raspberry Pi 3B+'s I am getting a intermittent results when calling a script with a button push. Everything works fine on two out of the three 3B+'s. Button pushes are not detected. I am using the same script, same Pijuice hat, and same execute permissions on my script on the three devices. The only difference I notice is that the devices where the button press is detected run pijuice_sys.py as root where the device that doesn't is running pijuice_sys.py as user pijuice. All settings in the PiJuice settings are the same on each device. The device that isn't working had a sudo apt-get upgrade issued prior to installing the pijuice-gui where the others did not. It's not clear if that is related in anyway.

tvoverbeek, 2020-01-08:
pijuice_sys.py is the pijuice service started by systemctl and should be running as root. 
This is the expected output of `systemctl status pijuice`:
```
pi@raspberrypi:~ $ systemctl status pijuice
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-01-07 21:55:59 CET; 15h ago
 Main PID: 573 (pijuice_sys.py)
    Tasks: 1 (limit: 4035)
   Memory: 9.3M
   CGroup: /system.slice/pijuice.service
           └─573 /usr/bin/python3 /usr/bin/pijuice_sys.py

Jan 07 21:55:59 raspberrypi systemd[1]: Started PiJuice status service.
pi@raspberrypi:~ $ 
```
The contents of the service file `/lib/systemd/system/pijuice.service`:
```
[Unit]
Description=PiJuice status service
After=network.target

[Service]
Type=idle
User=pijuice
WorkingDirectory=/var/lib/pijuice/
ExecStart=/usr/bin/pijuice_sys.py
ExecStopPost=/usr/bin/pijuice_sys.py stop
Restart=always

[Install]
WantedBy=multi-user.target
```
Do not know how you got it running as user pijuice. Did you start it manually?

thomerc1, 2020-01-08:
The service is active (running). Unfortunately, I did not start anything manually to arrive at this behavior. After a reboot, when I issue a "ps aux | grep python", I see that /usr/bin/pijuice_sys.py is running as user pijuice. I've since changed several pijuice files under /usr/bin to owner and group root with rwx privileges. The pijuice_cli, pijuice_cli.py, pijuice_gui, pijuice_gui.py were user and group pijuice and did not have execute privileges. Upon reboot, pijuice_sys.py is still running under user pijuice. When I stop the pijuice process and and manually run pijuice_sys.py as root, button presses are detected as expected. Also, my pijuice.service file is as the example you showed. Now that all files are owned and in group root should I change the user in the pijuice.service file to root?

tvoverbeek, 2020-01-08:
I was wrong. pijuice service should run as user pijuice.
Here is my output from `ps -ef | grep pijuice`:
```
pi@raspberrypi:~ $ ps -ef|grep pijuice
pijuice    570     1  0 15:50 ?        00:00:13 /usr/bin/python3 /usr/bin/pijuice_sys.py
pi         778     1  0 15:50 ?        00:00:22 /usr/bin/python3 /usr/bin/pijuice_tray.py
pijuice  21621     1  4 19:22 ?        00:00:00 /usr/bin/pijuice_gui /usr/bin/pijuice_gui.py
pi       21718 21304  0 19:22 pts/0    00:00:00 grep --color=auto pijuice
pi@raspberrypi:~ $ 
```
Both the service and the gui are running as user pijuice.
Note: the current version uses python3 not python2.
Does the pijuice user belong to group i2c, so it has access to the i2c bus?
Look at the entry for i2c in /etc/group.
Mine looks like `i2c:x:998:pi,pijuice`

thomerc1, 2020-01-08:
The pijuice user does belong to the i2c group and python3 is called to run the .py files. I placed a print statement under line 325 of pijuice_sys.py to print the variable ret. I see that I'm able to detect when the charger is inserted or removed - just am not able to detect the button pushes.

tvoverbeek, 2020-01-09:
How do you want to detect the button pushes?
You can add some print statements in _EvalButtonEvents and then run pijice_sys.py in a terminal (`sudo -u pijuice python3 <your modifued pijuice_sys.py>` after first having stopped the pijuice service).
Note, that with the service running, the service 'eats' all button events and you will not see any in a user python script.

thomerc1, 2020-01-09:
Prior to running the pijuice_sys.py, I stopped the pijuice service. The ret variable at line 325 is ret = pijuice.status.GetStatus(), which appears to return a string representing all detectable events. I was printing this every loop to detect events. There is a Boolean value 'isButton' that is in the string that is True when a button event is detected. I was unable to detect button events but can see when other events occur. However, I can detect button events when running as root. Any other config files you can think of that I maybe need to give execute privileges to?

tvoverbeek, 2020-01-09:
Button config is stored inside the PiJuice.
Since i2c communication with the PiJuice seems to work for everything else  I have no explanation for what you are seeing.
The only thing I can imagine is there is an other pijuice.py module in one of your directories which is picked up when running as normal user and not as root (or vice versa).
An other possibility is a damaged firmware. Try to reload the firmware.

andrewsmyname, 2020-01-10:
I'm having the same issue, my PiJuice was responding to button pressed and running user scripts just fine until I got the PaPiRus HAT. 

Besides installing that HAT, the only other thing I did was change the PiJuice EEPROM address to it's alternate value, for compatibility with the PaPiRus.

The PiJuice continues to be fully functional in every other way, and the service is running fine. I'm on Raspbian Buster on a Pi4.

andrewsmyname, 2020-01-20:
I tried your suggestion of adding some print statements to the pijuice service and running it manually, when I press **SW3** which is configured to `SINGLE PRESS: SYS_FUNC_HALT_POW_OFF, 0`, the python service runs `_EvalButtonEvents` and then ends up returning `true`, but nothing else happens.

If I press **SW2**, which is configured to `SINGLE PRESS: USER_FUNC 1, 400`, then `_EvalButtonEvents` doesn't seem to run at all as nothing is printed out whatsoever. Here's the minor changes I had made:

```bash
def _EvalButtonEvents():
    print("attempting to run evalbuttonevents")
    btEvents = pijuice.status.GetButtonEvents()
    if btEvents['error'] == 'NO_ERROR':
        for b in pijuice.config.buttons:
            ev = btEvents['data'][b]
            if ev != 'NO_EVENT':
                if btConfig[b][ev]['function'] != 'USER_EVENT':
                    pijuice.status.AcceptButtonEvent(b)
                    if btConfig[b][ev]['function'] != 'NO_FUNC':
                        ExecuteFunc(btConfig[b][ev]['function'], ev, b)
        print("returning true")
        return True
    else:
        print("returning false")
        return False
```

The PiJuice had worked flawlessly, the buttons etc working great, running my custom user script, and at some point after getting the PaPiRUS HAT they stopped working and haven't worked since. As I said, the only changes I made were to add that eink HAT to the system, and to change the PiJuice's EEPROM address to the alternate for compatibility.

If you need me to run any more tests or check anything else out let me know, because I'd love to get the PiJuice working again, thank you so much for the help @tvoverbeek - you can also reach me on the Pi Supply Discord :-)

As it is, it's not just the buttons that don't work - when the system is disconnected from power the UPS should immediately run my user script to shut the system down (which it used to do), but now it just sits there draining down the battery! :(

andrewsmyname, 2020-01-21:
Again, I've put a few more print statements in that function in case it's of any help, here's the results when I press SW3 (it doesn't register my SW2 press at all, as previously mentioned - and as previously mentioned SW3 does not trigger the configured response at all).

```bash
# Running pijuice_sys.py with the additional print statements:
attempting to run evalbuttonevents
btEvents =
{'data': {'SW1': 'NO_EVENT', 'SW2': 'NO_EVENT', 'SW3': 'PRESS'}, 'error': 'NO_ERROR'}
b =
SW1
ev =
NO_EVENT
b =
SW2
ev =
NO_EVENT
b =
SW3
ev =
PRESS
returning true
attempting to run evalbuttonevents
btEvents =
{'data': {'SW1': 'NO_EVENT', 'SW2': 'NO_EVENT', 'SW3': 'RELEASE'}, 'error': 'NO_ERROR'}
b =
SW1
ev =
NO_EVENT
b =
SW2
ev =
NO_EVENT
b =
SW3
ev =
RELEASE
returning true
```

```bash
def _EvalButtonEvents():
    #print("attempting to run evalbuttonevents")
    btEvents = pijuice.status.GetButtonEvents()
    #print('btEvents = ')
    #print(btEvents)
    if btEvents['error'] == 'NO_ERROR':
        for b in pijuice.config.buttons:
            #print('b = ')
            #print(b)
            ev = btEvents['data'][b]
            #print('ev = ')
            #print(ev)
            if ev != 'NO_EVENT':
                if btConfig[b][ev]['function'] != 'USER_EVENT':
                    pijuice.status.AcceptButtonEvent(b)
                    if btConfig[b][ev]['function'] != 'NO_FUNC':
                        ExecuteFunc(btConfig[b][ev]['function'], ev, b)
        #print("returning true")
        return True
    else:
        #print("returning false")
        return False
```

tvoverbeek, 2020-01-21:
@andrewsmyname From your log you can see SW3 is recognized and generated a press and release event.
You want to react on single press which you have configured with a 0 parameter. For single press you
need a non-zero parameter (indicates length of press in msec).
In your SW2 config you have configured USER_FUNC 1 (with space). No spaces here, should be USER_FUNC1. Typo?
You could add another print statement just before the `if btConfig[b][ev]['function'] != 'USER_EVENT':` line: `print(btConfig[b][ev]['function']` to see which function gets called.

andrewsmyname, 2020-01-21:
Ah! There's a difference between `PRESS` and `SINGLE PRESS`? I see! My bad on that one, sorry.

Also yes, it was a typo my apologies, `pijuice_cli` shows the correct value - `SINGLE_PRESS: USER_FUNC1, 400` .

I've added the print statement you asked for, as well as ensured that I've got `USER_FUNC1` under `PRESS`, and here's the result when pressing SW3:

```bash
USER_FUNC1
sudo: unable to execute /home/pi/scripts/pijuice-shutdown.sh: No such file or directory
```

That file exists by the way, I checked to ensure there's no unusual typo that I'm just not spotting by copying and pasting that output directly and looking at the file. The file's also got the correct permissions to allow any user to execute it (it had been working previously, nothing's changed with that file).

Also, do you have any way to troubleshoot why SW3 seems to be detected and is (at least) attempting to trigger the correct function, but SW1 and SW2 are not being registered whatsoever any more?

Thanks again for the help :-)

tvoverbeek, 2020-01-21:
Make sure your script is executable and owned by user pi, not by root.
First line of the script should be `#!/usr/bin/python3`

andrewsmyname, 2020-01-21:
Quick update @tvoverbeek , Ryan just helped me out with that error message, in the PiJuice docs the shebang line it asks you to use is `#!/usr/bin/bash` but that doesn't exist on my Pi, changing it to `#!/usr/bin/env bash' now allows the user script to run.

So! There's problem one solved! Two remain:

1. The system doesn't trigger that same script when power is lost, even though it's set to do so
2. SW3 is detected, but SW1 and SW2 are not.

tvoverbeek, 2020-01-21:
What is the output of `ls -l /home/pi/scripts/pijuice-shutdown.sh` and `grep pijuice /etc/group`?

andrewsmyname, 2020-01-21:
```bash
pi@calipi:~ $ ls -l /home/pi/scripts/pijuice-shutdown.sh
-rwxrwxrwx 1 pi pi 553 Jan 21 20:21 /home/pi/scripts/pijuice-shutdown.sh
pi@calipi:~ $ grep pijuice /etc/group
i2c:x:998:pi,pijuice
pijuice:x:1001:pi
```

tvoverbeek, 2020-01-21:
That all looks OK. Have you checked the no power setting on the system events tab and the correct assignment of USER_FUNCx on the User scripts tab?
If all these are OK it should work.

andrewsmyname, 2020-01-21:
[image]

Yes sir, it's all set up :-) Like I say, it was working fine when I first got the PiJuice and got it all configured, and then after I set the PaPiRUS up and went back to test everything out again, that's when I found out that the PiJuice was having these problems.

andrewsmyname, 2020-01-21:
Quick update, I was playing around a bit, when the service is running by itself it doesn't run my custom script for whatever reason, but when I run it manually (as the pijuice user) it functions correctly and runs it, which shuts the system down. e.g with some print statements:

```bash
running evalpowerinputs
1
running evalpowerinputs
2
unplugged
2
```

Weird issue though, I'm sat here right now after the above happening, and the UPS is continuing to power the Pi even though the Pi has been entirely shut down (pressing SW1 does boot the Pi right back up, but yeah, that's not the right behaviour, the UPS should power down shortly after the Pi shuts down).

I hope my mucking about gives you some ideas :-)

tvoverbeek, 2020-01-22:
The service should also run as user pijuice (defined in the service file: /lib/systemd/system/pijuice.service
In the [Service] section there should be a line `User=pijuice`

Check with `ps -ef | grep pijuice_sys`.
My output is: `pijuice    563     1  0 16:16 ?        00:00:01 /usr/bin/python3 /usr/bin/pijuice_sys.py`
First item is the user under which the process runs.

andrewsmyname, 2020-01-22:
My output is the same.

So just to reiterate so we're on the same page *(because I edited some of my earlier comments, hope thats okay)* my only issue now is that when I press `SW2`, the PiJuice doesn't seem to notice (_EvalButtonEvents is not triggered at all).

By the way, thanks for the help :-) Once this is all sorted if you're ever in Manchester I'll buy you a pint! :-)

tvoverbeek, 2020-01-22:
@andrewsmyname Do not know if I make it to Manchester, but if I do we'll take a pint.
Just for the sake of completeness, if you remove the PaPiRus (when not too difficult) does SW2 work then? When removing PaPiRus disable the epd-fuse service.

andrewsmyname, 2020-01-22:
PaPiRus removed, all three buttons on the PiJuice are set to run `USER_FUNC1` _(herein called the script)_ on `PRESS`, and here are my results:

**SW1 & SW3:** Runs the script fine every time I press them, no problems.
**SW2**: Does nothing whatsoever.

I'm starting to wonder if perhaps it's a hardware fault, and the button itself is faulty? There is nothing amiss with a close visual inspection, and the push "feels" fine.

andrewsmyname, 2020-01-28:
**Small Update:** Checked the physical button with the multi-meter and it functions fine, so not a hardware issue.

shawaj, 2020-10-24:
@andrewsmyname did you solve this issue in the end?

@thomerc1 need any other help?