PiSupply/PiJuice #442: command line scripts or examples

jdimpson, 2020-01-13:
Feature request: I'd love to see some command line scripts or examples of using your api from command line.

(Its pedantry, but the pijuice_cli.py program is not command line, its a tui, or text user interface. A cli program reads arguments from the command line arguments, and prints results to standard out, and in general is suitable for scripting.)

In particular, I'd like examples of grabbing the battery percentage and of installing or removing user functions from the command line, e.g.

pi@raspberrypi:~ $ pijuice_battery.py
Battery percent: 79

pi@raspberrypi:~ $ pijuice_install_script.py USER_FUNC1 /home/pi/myjuicescript.sh

Ive looked at the code but I'm not sure I have the skills to do this myself, but if i figure it out ill submit a patch, or however you do that in git.

shawaj, 2020-10-24:
@jdimpson does the script in #518 help you at all?

Also see #88 and #496

Think it might be what you're looking for

jdimpson, 2020-10-24:
It sure does! I can't actually remember what I had in mind because I ended up using my pijuice with the media hat (so the gui command is the right solution there).  But the addition of pijuice_util.py is the kind of thing I had in mind, and also provides a good simple example of using the API.

closed 2020-10-24 by jdimpson

shawaj, 2020-10-24:
@jdimpson perfect 👍