PiSupply/PiJuice #445: Communication error when running on battery

Elwetana, 2020-01-18:
I am using PiJuice Hat with firmware 1.3 and stock BP7X battery and Pi A+. Everything works fine, but when I disconnect the external power, after a short while I am unable to communicate with PiJuice: I get CONNECTION_ERROR.

The weird thing about this is that id does not happen immediately, only some 20-30 seconds after I disconnect the power. The Pi itself runs ok on battery and when I reconnect the power, everything works again.

I am using fairly basic Python script to communicate with PiJuice.

Is this expected behaviour?

tvoverbeek, 2020-01-19:
Is this a Pi 1 A+ or a Pi 3 A+ (difference between single core and multi core)?

Elwetana, 2020-01-19:
It's PI 3 A+.

tvoverbeek, 2020-01-20:
@Elwetana  Pi 3A+ varies its system clock depending on the cpu load. The i2c clock frequency changes with the system clock. At light CPU load the i2c clock is about half of what is specified (default spec = 100kHz). When the i2c clock frequency changes in the middle of a transaction you get problems and you can end up in a continuous COMMUNICATION ERROR.

As a test try the follwoing:
In a terminal window give the command `cat /dev/urandom > /dev/null &`.
This will load one core at 100% (CPU load will show as 25%).
You can start up four ot these to load all 4 cores.
Because of the extra load the system clock will go to max and the i2c clock to a stable 100kHz.

Try first with loading a single core before you disconnect external power. See if the COMMUNICATION ERRORs do not occur.
Let us know how it goes.

Elwetana, 2020-01-20:
@tvoverbeek Thank you, that was a really good insight. I did as you suggested and I am not getting the error anymore. So what now? When I lose the power, it's probably not a good to idea to increase power consumption just to keep the cores spinning. Should I try setting the I2C frequency to some lower value? Also, I could only run some power-hungry process when I am about to query PiJuice (I am only doing it every five minutes).

At any rate, many thanks for considering this problem.

tvoverbeek, 2020-01-20:
@Elwetana The simplest and most power saving option is to fix the GPU core frequency to 250MHz,
Add the following 2 lines to /boot/config.txt:
```
core_freq=250
core_freq_min=250
```
This prevents the i2c clock frequency from changing. It will stay at its lowest frequency.
Try it and see if it continues working.

Elwetana, 2020-01-20:
@tvoverbeek I've followed your advice and updated my script to check PiJuice every 30 seconds. The results are inconclusive -- in about half the cases I get the communication error (with power disconnected).

When I increase the load again (by engagin urandom as per your suggestion) it seems to work without the communication errors.

tvoverbeek, 2020-01-21:
@Elwetana OK, so the low i2c clock frequency is probably on the edge.
Could you try to replace the `core_freq=250` and `core_freq_min=250` lines with
`force_turbo=1` and reboot.
This fixes the system clock on the highest frequency and the i2c clock to the default 100kHz.
Will use a bit more power, but much less as the extra load.

Elwetana, 2020-01-21:
@tvoverbeek This sounds very reasonable and I am really grateful for your suggestions. Unfortunately, the result is very similar to previous case -- about 50% of the time I get the error. Do you have some other idea? I am now thinking about spawning another process before reading the status, let it stress the CPU for a bit, then reading the status and joining the process.

tvoverbeek, 2020-01-22:
@Elwetana Since you are on firmware 1.3 I have one last suggestion.
With 1.3 on the Pi4 we need to use the old i2c driver (fixed in the upcoming release). You could try the same for your 3A+.
Put the following line in /boot/config.txt: `dtoverlay=i2c-bcm2708`. After reboot check with lsmod that i2c_bcm2708 has been loaded and not i2c_bcm2835.
Suggest to remove the `force_turbo=1` first and add it later again if this does not work.

Elwetana, 2020-01-23:
@tvoverbeek I am afraid this didn't make much difference either. It seems that I get the communication error slightly less often, but I think that's just statistical noise. It's also interesting that the error never occurs when the Pi is powered (I am powering Pi directly, not via PiJuice). I would be very happy to share more information with you if you think it could be useful, and I can work around the problem (I will just retry the connection few times). Many thanks for your suggestions anyway. Feel free to close this issue if you think there's nothing more to be done.

tvoverbeek, 2020-01-24:
The only other thing I can think off is that the BP7X does not have sufficient capacity to drive the A3+ with peripherals.
Closing the issue anyhow.

closed 2020-01-24 by tvoverbeek