PiSupply/PiJuice #446: NO POWER event does not run my function

gridtechewz, 2020-01-20:
Hi everyone,

I have written a simple code, which sends an email to a person. When I run it manually, it works. So, the code should be fine.
I have added this code to USER FUNC1 and when "No power" happens, it will trigger the USER_FUNC1.
Unfortunately, if I unplug the power cable and the RPI4 runs via battery, it does not run the USER_FUNC1...

Also, this code should be run, when I start the RPI4 via crontab. This also does not work. 

Could someone explain me, what went wrong? In the following, you see my settings and crontab entry.

[image]
[image]
[image]

tvoverbeek, 2020-01-21:
Your script Ausschaltmeldung.py should be executable and owned by user pi.
First line of script should be `#!/usr/bin/python3`.
The crontab job will run not at shutdown but as part of the reboot.
It runs very early in the boot sequence, most likely before the i2c is initialized and the network is up.

gridtechewz, 2020-01-22:
> Your script Ausschaltmeldung.py should be executable and owned by user pi.
> First line of script should be `#!/usr/bin/python3`.
> The crontab job will run not at shutdown but as part of the reboot.
> It [...]

Ok, thanks for the advice! Now, the script works, when I unplug the power cable!
What would you advice me to do, in order to use a script, 
1.  when I turn on the RPI?
2.  when I plug the power cable back?

tvoverbeek, 2020-01-22:
@gridtechewz 
For 1)
call your script from /etc/rc.local. But you will have to add a test for network connectivity first before calling your Ausschaltmeldung (Einschaltmeldung???)
For 2)
Enable wakeup-on-charge with trigger level 0.
However the firmware clears its internal wakeup-on-charge flag when it starts the Pi.
So somewhere you will have to enable it again in your script(s): `pijuice.power.SetWakeUpOnCharge(0)`.

gridtechewz, 2020-01-28:
> @gridtechewz
> For 1)
> call your script from /etc/rc.local. But you will have to add a test for network connectivity first before calling your Ausschaltmeldung (Einschaltmeldung???)
> For 2)
> Enable [...]

Thanks, I could solve problem 1.) with your answer, but I don't get your answer from 2.)
I could change via the PiJuice Settings: Wakeup on charge to 0 and I also enabled Wakeup, but I don't see, how it calls by script "Einschaltmeldung.py" though. Could you elaborate it more?

gridtechewz, 2020-01-29:
> @gridtechewz
> For 1)
> call your script from /etc/rc.local. But you will have to add a test for network connectivity first before calling your Ausschaltmeldung (Einschaltmeldung???)
> For 2)
> Enable [...]

Hey, I finally solved problem 2! Thanks for the help!

closed 2020-01-29 by gridtechewz