PiSupply/PiJuice #449: PIjuice and Pi like a phone

bsuarez49, 2020-01-21:
Hello,
I am trying to configure PiJuice to work similarly (or almost) to a phone:

If the PI is powered by a USB power supply:
- the system must remain on whatever the level of the battery

If USB power is disconnected:
- the PI uses the pijuice battery
- when the battery becomes weak the PI must stop properly.

If I reconnect the USB power supply:
- the system must restart regardless of the battery level

If I shut down the system with the command "shutdown" or through the raspbian menu:
- the system must remain off even if the power supply remains connected
- Disconnecting and reconnecting the power supply must restart the system.

I tried several configurations but I can't seem to have this working. Currently my configuration is as follows:

System Task:
- Wakeup on charge: 0%
- Minimum charge: 20%
System Events:
Low Charge: SYS_FUNC_HALT_POW_OFF

In this case, the system stops well if the battery becomes less than 20% but if I reconnect the power later, the system restarts but turns off immediately because the battery is still below 20% (loop as long as the battery is <20%) . In addition if I use the "shutdow" command with USB power connected, the PI turns off but restarts because the power is on.

Is the scenario I want possible? If so, can you guide me for the configuration.

Thank you

tvoverbeek, 2020-01-21:
I believe you can achieve what you want this way:
-Power is only applied via the Rpi usb, not the PiJuice usb connector.
-Disable wakeup on charge completely.
-Leave Minimum charge at 20% but attach the Low Charge to a userfunction, e.g. USER_FUNC1
In the user function check if power is connected (use pj.status.GetStatus and check powerInput and powerInput5vIo) If both power inputs are false shutdown but set the poweroff to 20 sec first.
-On the System task tab enable the Software Halt Power Off with a delay period of 20 sec.

This will disable the 5V from the PiJuice to the RPi after shutdown and put the PiJuice in low power mode, minimizing battery drain.
Reapplying power should then restart the RPi and since the user function will see at least one power connection there will be no immediate shutdown.

bsuarez49, 2020-01-23:
Hello,
Thanks for your reply, I will try a script when I have some time. I thought the software planned this use case!
For now I have disabled the Wakeup_on_charge and the Minimum Charge and I have configured the Sofware_Halt_Power_Off to 30s and in System_Events only attached No_power to SYS_FUNCT_HALT_POW_OFF.
So if I unplug the power supply (Pi USB) Pijuice does its job and the card turns off cleanly.

shawaj, 2020-02-22:
@bsuarez49 did you have any further questions here?

Arno-GitHub, 2020-06-08:
I do have a question. I'm trying to do the same with power only connected to the RPi4 and a charged battery. Tried it with wakeup on charge 0% enabled. Using firmware 1.4 with OS buster.

What triggers the RPi to switch on?
The battery is charged (>70%) so I would expect there is no 'Low charge' event and the function or script is not executed.

@tvoverbeek: Why do you suggest to disable wakeup on charge completely? I would think that is the trigger to power on the RPi when connecting USB power to the RPi.
And here (https://github.com/PiSupply/PiJuice/issues/446) you suggest to enable wakeup on charge. Can you please explain?

Reading issue #446 should I enable wakeup on charge and execute 'pijuice.power.SetWakeUpOnCharge(0)' in a python3 script in /etc/rc.local? Both do the same isn't it?

Arno-GitHub, 2020-06-08:
Unfortunately this doesn't power up the Pi when connecting USB power to the Pi.
Any suggestions?

Arno-GitHub, 2020-06-09:
Nothing changed in the settings, different outcome.
Shutdown in PiOS, Pi did not shutdown but rebooted.

tvoverbeek, 2020-06-09:
@Arno-GitHub Need to know your PiJuice setup details and any user functions you used.
The setup I outlined on Jan 21 to @bsuarez49 treis to make sure the 5V from the PiJuice is off after every shutdown (both programmed and from the OS). Only then will reconnecting power to the Pi usb start the boot.

If your Pi rebooted instead of shutting down then you probably have wakeuponcharge enabled.
Are your requirements exeactly the same as the ones in the first post?
If not, specify what you want to acheve.

Arno-GitHub, 2020-06-09:
Yes I have wakeuponcharge enabled because I can see that's the only way to trigger the Pi to switch on.
Next I will test it with wakeuponcharge disabled and no scripts in /etc/rc.local.
My requirements are:
- Power off Pi in OS (Shutdown - Shutdown in GUI)
- Power on Pi by plug in USB power on Pi regardless of battery charge.
- When Pi is on and the battery is charged, remain on on battery power when power to USB on the Pi is unplugged.
- Bonus: Shutdown PiOS properly when only on low battery charge.

Same as above I believe.
Tricky one is I can't access the PiJuice without unscrewing the backcover of the case (SmartiPi Touch 2).

Arno-GitHub, 2020-06-09:
wakeuponcharge disabled and no pijuice scripts in /etc/rc.local
Shutdown twice. First time the Pi rebooted. The second time the Pi shutdown but didn't power on when I disconnected and reconnected the USB power to the Pi.

So I had to unscrew the backcover, push a button on the PiJuice (don't know which one, has to be sw1 if I see the HAT configuration)

Next I will test 'pijuice.power.SetWakeUpOnCharge(0)' with wakeuponcharge disabled.

Arno-GitHub, 2020-06-09:
'pijuice.power.SetWakeUpOnCharge(0)' in /etc/rc.local with wakeuponcharge disabled.

PiOS: Shutdown - Shutdown powers off the Pi but not powers on when USB power is reconnected.
I think that is expected because /etc/rc.local is not executed.

PiOS: Shutdown - Shutdown second time reboots the Pi.

tvoverbeek, 2020-06-10:
I assume you are powering the Pi+PiJuice via the USB-C of the Pi4.
In the PiJuice Settings System Task Tab: System Task enabled, Wakeup On Charge disabled,
Minimum Charge enabled and set to e.g. 15%, Software Halt Power Off enabled and timeout
set to min value(20 sec).
On the System Event Tab: Low Charge enabled with SYS_FUNC_HALT_POWER_OFF.

This would ensure that the 5V from the Pijuice to the Pi gets switched off when you shut down the Pi, either from PiOS of by Low charge. Then a USB-C power cycle reboots the Pi.

However in the current 1.4 firmware there is one caveat: **When the 5V to the PiJuice is coming back or stays present during the delay (20 sec for PiOs shutdown, 30 sec for SYS_FUNC_HALT_POWER_OFF) then the PiJuice 5V to the Pi is not switched off.**
If you then powercycle the USB-C power the Pi will not reboot, since it is still has 5V from the PiJuice.

Make sure the USB-C 5V is removed before you shut down the Pi from PiOs (For low charge I assume it is removed already since you are on battery power only).

Note that you can connect an external push button to J5 on the PiJuice which is in parallel to SW1.
Then you have a possibility to reboot the Pi in any case.