PiSupply/PiJuice #450: PiJuice Firmware broken after changing I2C address 14 and flashing

uGiFarukh, 2020-01-23:
Hello,

Recently I have been wanting to change the RTC I2C address of the PiJuice HAT because it was conflicting with my MPU6050 I2C address 0x68. After changing the RTC address everything worked like charm until I tried to change the 0x14 I2C address of the PiJuice for testing. The address changed successfully but both the pijuice_cli and pijuice_gui was giving communication error. Then I tried to flash V1.3 firmware using pijuiceboot but then it failed the flash with EEPROM write error (something like that). And now the pijuice is totally bricked. No LEDs inside the pijuice is running and no buttons are working.

How can I solve this bricked firmware issue without removing the PiJuice BP7X battery? The pijuice battery is in a very difficult place to be removed inside the the project. I have to undo lots of things if I have to get to the battery. Is there any solution without removing the battery? Currently both the pijuice and raspberry pi is in off state. No blinking leds.

A solution to this would be highly appreciated. Thanks.

tvoverbeek, 2020-01-23:
Did you use the changed address as argument with the pijuiceboot command?
Note the address argument is in hex (without any leading '0x').
But otherwise to recover you need access to the PiJuice Buttons and the battery.
See https://github.com/PiSupply/PiJuice/tree/master/Firmware#faq

uGiFarukh, 2020-01-23:
Yeah I have used the hex address (without any leading 0x) as the argument on the pijuiceboot. Alright I will try to reach the battery for solving this.

closed 2020-01-23 by uGiFarukh