PiSupply/PiJuice #452: can the Pijuice power a pi4 and the Official screen

Calgar77, 2020-01-26:


shawaj, 2020-01-26:
@Calgar77 yes it should work, as long as you use the 12000mAh battery to give maximum output

Calgar77, 2020-01-26:
ok thank you

shawaj, 2020-02-22:
Did you manage to get this working @Calgar77 ?

Calgar77, 2020-02-23:
unfortunatly i have been unable to test it due to the battery not fitting into the case i'm using, so am currently looking into custom modding the case to fit it .

Arno-GitHub, 2020-06-08:
At the moment i'm using the Official 7" touchscreen and the BP7X battery.
The Pijuice can then only be used as an UPS.

I'm using the BP7X because of the size of the case (SmartiPi Touch 2).
After ~30 minutes the battery is drained.

shawaj, 2020-10-24:
@Calgar77 did you test this now?

@adevoss is that with the pi 4 and official display? It works ok and doesn't give low voltage warning with the BP7X?

adevoss, 2020-10-24:
@shawaj Yes it is the official display. It does not work ok. I got help of a Pijuice developer. He said the display draws too much from the battery.

I couldn't get it to work so I removed the Pijuice and now using my Pi with a power supply.
With that power supply I get low voltage warnings from time to time.

shawaj, 2020-10-24:
@adevoss you could always try the 12000mAh battery but that won't fit so nicely in the case.

adevoss, 2020-10-24:
@shawaj Does not fit indeed. I want to use it on the move (satnav in the car). Haven't found a solution that works for me.
When I use the cigarette lighter there are so many cables.
At shutdown the touchscreen goes off immediately but the OS is not shutdown yet. Also not helps.

So I shelved that project a while ago.
Use the Pi now as my desktop pc. Until the Intel NUC 11 will be released. Then the Pi will be available for a next project.
Thinking about Kodi or satnav (navi) again if other solutions are available.

shawaj, 2020-10-24:
@adevoss something like this might be useful https://www.amazon.co.uk/dp/B07P87KXNY/

adevoss, 2020-10-24:
@shawaj Thanks. Those wires without a connector always scare me a bit. I don't want to be fried by my car battery. But I will look into it again.

shawaj, 2020-10-24:
@adevoss you could use some connectors like this to hook them up to the battery https://www.amazon.co.uk/dp/B0787B99KB/

You could add an inline fuse if you want too https://www.amazon.co.uk/dp/B076CFS2M6/

A 1.5A or 2A fuse should be good.

The 12 volts of the car battery won't necessarily give you a shock, unless your hands are very wet, as the resistance of human skin doesn't allow much current to flow at 12v. So you don't need to be too worried when working on a12v system.

shawaj, 2020-12-05:
@adevoss  can we close this issue now or do you need any further assistance?

adevoss, 2020-12-10:
I did not start this issue so I can't say. I have no further questions.

shawaj, 2020-12-10:
Closing this issue for now. Please reply back if you disagree or have any further issues.

closed 2020-12-10 by shawaj