PiSupply/PiJuice #454: Feature Requests

victorromeo, 2020-01-31:
I've got  a wish list for new features to add to the PiJuice api please.

- Setting up the PiJuice via a JSON blob, to restore from source control
- Scripts on IO1 and IO2 pins rising and falling edges
- Option for a 5V peripheral supply while Pi is Off
- Built in algorithm to determine an approximate time remaining
- Built in lifecycle assistant to measure when batteries are needing replacing
- Pub/Sub events on the IO1 and IO2 pins, similar to gpiozero callback events
- Inline USB protection to prevent killing a Pi (A piece which adapts onto the USB port)
- GPIO pin protection
- A way to switch off the camera via the PiJuice, (Camera plugs into PiJuice, PiJuice regulates and switches the camera on and off)
- Software re settable fuses
- Headless options for the Pi4, to slim the unit down

Food for thought

ChristopherRush, 2020-01-31:
With regards to the last option we have - https://uk.pi-supply.com/products/pijuice-zero

shawaj, 2020-10-24:
Closing this as it's a duplicate of #451

closed 2020-10-24 by shawaj