PiSupply/PiJuice #455: Setting a maximum charge percentage/level?

jbmorris289, 2020-02-01:
Hi! Let me get straight into this.

I use my PiJuice mainly as a UPS for my Pi (that will change soon), so it is (almost) always plugged in. I'm pretty sure a lot of us have heard that keeping your battery fully charged will shorten the lifespan of the battery. Also, according to some things I've read, batteries are happiest at percentages around 40-70% - or something like that. 

I want to care for my battery, so my question is... is there a way to set a max charge level somehow?

galinette2000, 2020-02-01:
This would be for sure an interesting feature included in the service and the GUI, as default behavior will kill the battery in a UPS use. This is a +++ feature request.

I'm trying to do it via API with this script. However, disabling the charge does not work for unknown reasons, I've submitted an issue for this.

```
#!/usr/bin/python3 
from pijuice import PiJuice # Import pijuice module 

charge_high = 70
charge_low = 60
pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object 
status = pijuice.status.GetStatus() 
charge = pijuice.status.GetChargeLevel() 
charging_config = pijuice.config.GetChargingConfig() 
print(charging_config) 
print(charge) 
print(status) 
if charge['error']=='NO_ERROR' and status['error']=='NO_ERROR' and charging_config['error']=='NO_ERROR': 
    status = status['data'] 
    charging_config = charging_config['data'] 
    charging_enabled = charging_config['charging_enabled'] 
    powered = status['powerInput']=='PRESENT' or status['powerInput5vIo']=='PRESENT' 
    level = charge['data'] 
 
    if powered: 
        print('Board is powered') 
    else: 
        print('Board is not powered') 
 
    if charging_enabled: 
        print('Battery is charging') 
    else: 
        print('Board is not charging') 
 
    print('Battery level is '+str(level)+'%') 
 
    if status['battery'].startswith('CHARGING_FROM') and level>=charge_high and charging_config['charging_enabled']: 
        print("Disabling charge") 
        pijuice.config.SetChargingConfig(False, True) 
        print(pijuice.config.GetChargingConfig()) 
    elif status['battery'].startswith('NORMAL') and level<charge_low and not charging_config['charging_enabled']: 
        pijuice.config.SetChargingConfig(True, True) 
        print("Enabling charge")
```

galinette2000, 2020-02-01:
Found the issue in the script above! It works now.

tvoverbeek, 2020-02-04:
See Issue #456 . Closing this issue.

closed 2020-02-04 by tvoverbeek

ph818, 2023-02-06:
Thank you for this script. I found that I had to add in some closing } for it to work for me. I didn't try the more 'natural' syntax.

```
#!/usr/bin/python3 
from pijuice import PiJuice # Import pijuice module 

charge_high = 70
charge_low = 60
pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object 
status = pijuice.status.GetStatus() 
charge = pijuice.status.GetChargeLevel() 
charging_config = pijuice.config.GetChargingConfig() 
print(charging_config) 
print(charge) 
print(status) 
if charge['error']=='NO_ERROR' and status['error']=='NO_ERROR' and charging_config['error']=='NO_ERROR': 
    status = status['data'] 
    charging_config = charging_config['data'] 
    charging_enabled = charging_config['charging_enabled'] 
    powered = status['powerInput']=='PRESENT' or status['powerInput5vIo']=='PRESENT' 
    level = charge['data'] 
    if status['battery'].startswith('CHARGING_FROM') and level>=charge_high and charging_enabled: 
        print("Disabling charge") 
        pijuice.config.SetChargingConfig({'charging_enabled':False}, False) 
        print(pijuice.config.GetChargingConfig()) 
    elif status['battery'].startswith('NORMAL') and level<charge_low and not charging_enabled: 
        pijuice.config.SetChargingConfig({'charging_enabled':True}, True) 
        print("Enabling charge")