PiSupply/PiJuice #456: Unable to disable charge via API

galinette2000, 2020-02-01:
Hi,

With pijuice_cli, I can enable and disable battery charging without issues.
I want to write a script for controlling the battery charge and keep it between 60% and 70%, in order to preserve battery life as I use it always powered (UPS use)

I run in a python script with user pijuice (via sudo -u pijuice python3 test.py):
```
print("Disabling charge") 
pijuice.config.SetChargingConfig(False, True) 
print(pijuice.config.GetChargingConfig())

```
And I get:

> Disabling charge
> {'data': {'charging_enabled': True}, 'error': 'NO_ERROR', 'non_volatile': False}

And charge continues.

galinette2000, 2020-02-01:
I found the problem, now the command to enable charging is:
`pijuice.config.SetChargingConfig({'charging_enabled':True}, True) `

Instead of what is written in the documentation:
`pijuice.config.SetChargingConfig(True, True) `

tvoverbeek, 2020-02-02:
@galinette2000 Thanks a lot for catching this.
The `pijuice.config.SetChargingConfig({'charging_enabled':True}, True)` syntax is used by pijuice_cli and pijuice_gui.
When writing user scripts the syntax `pijuice.config.SetChargingConfig(False, True)` is more natural.
Will try to modify pijuice.py to accept both forms (so we do not have to modify the documentation ;-)).

tvoverbeek, 2020-02-03:
Done. Closing Issue.

closed 2020-10-24 by shawaj