PiSupply/PiJuice #457: Question about addressing the RTC

mhaserodt, 2020-02-05:
I am using a PiJuice Zero on top of a Zero W.  The Zero W has a camera connected to the CSI camera connector.  It is running MotionEyeOS in a headless/WiFi access point configuration.

With a similar setup without the PiJuice, I am able to use a ds3231 RTC with no trouble. The ds3231 is soldered into gpio pin 6 for ground, pin 2 for 5v power, pin 3 for SDA, and pin 5 for SCL.

MotionEyeOS has `dtparam=i2c_arm=on` added to the /boot/config.txt file as a default, and I have added `rtc-ds3231` to /data/etc/modules 

It requires a startup script /data/etc/userinit.sh which includes

```
echo ds3231 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
sleep 5
hwclock -s
```

With this configuration `i2cdetect -y 1` shows me UU at 68 and 14 at 14, so I can tell the RTC is being seen and loaded.  hwclock functions work fine and the pi reads and uses the RTC as it should.

I'm working with the PiJuice and having trouble getting the integrated RTC to work.  I've seen a little conflicting info where it seems the PiJuice RTC is addressed as either ds1307 or ds1339.  I've tried both to no avail.  I've tried various combinations using `rtc-ds1307` and `rtc-ds1339` in /data/etc/modules and both `echo ds1307...` and `echo ds1339...` in the userinit.sh.  I've also tried adding `dtoverlay=i2c-rtc,ds1307=1` and `dtoverlay=i2c-rtc,ds1339=1` to /boot/config.txt.  I've tried both HAT ID-EEPROM 0x50 and 0x52

Most of these permutations result in the userinit.sh script running and it returns 
'hwclock: can't open /dev/misc/rtc: no such file or directory'

which shows 68 at 68 instead of UU when running `i2cdetect -y 1`.  Of course the RTC can't be read or set then.

The closest I seem to get is running HAT ID-EEPROM 0x52, with `rtc-ds1307` in the userinit.sh, `echo ds1307...` in the modules, and `dtoverlay=i2c-rtc,ds1339=1` in config.txt.  With this combo it'll see and load the RTC (showing UU at 68) but when I try any hwclock command I get "RTC_RD_TIME: Remote I/O error"

Maybe this is an issue I need to address with MotionEyeOS, but I thought I'd start here first, so any thoughts on what I'm missing here?

mhaserodt, 2020-02-05:
I also forgot to add the PiJuice RTC works fine under Raspbian.  Both in the PiJuice setup menu and using hwclock commands in the command prompt.

tvoverbeek, 2020-02-05:
In general if you use a device-tree overlay you do not need to load the module manually and also the  `echo <rtctype> <addr> > ...` is not needed. All this is handled by the overlay.
I do not know if motioneyeos uses the same device tree overlays as Raspbian.
An other issue might be that you call your hwclock -s too early before the PiJuice has intialized its rtc i2c communication. Remember the PiJuice rtc is emulated in the firmware.
First try in a terminal (ssh window) `sudo hwclock -r --verbose`
Here is my output on Raspbian
```
pi@raspberrypi:~ $ sudo hwclock -r --verbose
hwclock from util-linux 2.33.1
System Time: 1580929270.120091
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
ioctl(3, RTC_UIE_ON, 0): Invalid argument
Waiting in loop for time from /dev/rtc0 to change
...got clock tick
Time read from Hardware Clock: 2020/02/05 19:01:11
Hw clock time : 2020/02/05 19:01:11 = 1580929271 seconds since 1969
Time since last adjustment is 1580929271 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2020-02-05 20:01:10.590318+01:00
pi@raspberrypi:~ $ 
```

shawaj, 2020-10-24:
@mhaserodt did you get this all sorted? Can we close the issue?