PiSupply/PiJuice #458: Slightly swollen battery..

jbmorris289, 2020-02-06:
Hi again!

So, I took a close look at the battery mounted on my PiJuice HAT, and it seems to be a little swollen. 

Before I posted another issue on here about battery charge regulating (max charge), my PiJuice Battery has been fully charged (well... at 98%) for a while. Could that have been the cause? (If yes, then... my fault for not closely monitoring.)

Should I consider getting another battery?

tvoverbeek, 2020-02-06:
@jbmorris289 Swollen battery is never good.
Probably gases have  been created inside. This is how exploding Li-Ion batteries start.
Time to contact the seller and get a replacement.
Do not continue using this battery.
Normally the PiJuice protects against overcharging. Moreover the BP7X also has a safety circuit.
But these batteries may be old. Mine are marked 2012.

jbmorris289, 2020-02-07:
@tvoverbeek Okay, I thought so. Will contact my seller. Would I able to remove it while my Pi is plugged in?

shawaj, 2020-02-07:
@jbmorris289 yes you can remove the battery whilst the unit is plugged into power without it shutting down.

I just replied to your support ticket too 👍

jbmorris289, 2020-02-07:
Okay, the above message has been replied to.

For reference, here is a picture of my PiJuice Battery. I probably should of provided this initially.
[image]

shawaj, 2020-02-07:
That's definitely not safe to use. Will close this issue now as we are dealing with it on the ticket

closed 2020-02-07 by shawaj