PiSupply/PiJuice #459: Battery Status is constantly changing

MattBond99, 2020-02-07:
I've just added PiJuice to my Raspberry PI 3B+ and noticed that the Battery Status on the Setting screen is constantly changing. It will  change about every 3-5 seconds going from a state of, say 98% - Charging from 5V... to 0% NOT Present  then back to CHARGING; but at a different charge level, say 82%.  This usually cycles constantly. Every now and then it seems to get into a steady state of CHARGING. As I type this its been in a steady state @ 80% - CHARGING for about 5 minutes.

Any Thoughts?

tvoverbeek, 2020-02-07:
Smells like bad contacts between the battery and the PiJuice.
If it is a PiJuice HAT with BP7X, have you removed the plastic piece between the battery and the connector? Maybe some bits were left behind.

MattBond99, 2020-02-07:
I did remove the plastic piece; but it came out  fairly easy; maybe the battery is not seated properly. I'll check when I get home today

shawaj, 2020-02-22:
It might be worth updating the software to v1.6 and the firmware to v1.4 as well and seeing if this improves things