PiSupply/PiJuice #460: Weird behaviors when powering via the Raspberry Pi USB port

rdragonrydr, 2020-02-07:
I have a Raspberry Pi 3b (non-Plus) attached to a PiJuice (non-Zero) hat that I have connected to a power supply via the Raspberry Pi's USB port.
___
Assorted things that may be important but are not part of the main issues:
Side note on something that may not be working correctly that I'd like advice on: On a few occasions, I've had to unplug the battery from the hat and reconnect it before it would be willing to charge. This would happen regardless of what power supply I was using, even my good one. Reconnecting the battery (which clearly is connected, since it can discharge just fine even when it won't charge) does make it work, though?

The charge/power LED also shines very brightly and constantly for a few seconds about every 5-7 minutes or so, rather than its usual very quick blink every few seconds. I do not know if it is supposed to do this.

Feature Requests: 
-Add a "power is on" LED mode to go with the "charge level" LED. It can be hard to tell if it's charging or actually on.
-Allow configuration of colors and levels for the charge LED. At least, mine never shows amber, only green or red (at <50%, which seems high).
___
1:) If the Pi does not have an SD card, but is powered via its USB, later inserting a card and pressing the poweron button has no effect (does not boot). Removing USB power (at which point it remains powered via the Hat) and pressing the button again DOES boot it. There's also a chance that it can become case 2 instead, and simply turn off, but only after power is removed (this has happened twice, and cannot be reliably replicated).
Result: I would expect this to still work if the Pi has USB power, since this is done via a GPIO event and not a power cycle. After all, when it does work with power disconnected, it's not like power to the Pi is turned off and then back on (that I know of). May drain the battery in the same manner as 2.
Note: it can start the Pi from a normal shutdown state even with the Pi's USB power connected, so I suspect some sort of desync where it thinks it's already on and waits for confirmation from the driver? Further note: Hat power remains on even if the Pi is not booted once power is disconnected.

2:) The hat may be manually powered on (conveniently, it does not automatically power on without the SD card) while connected to a USB-powered Pi that has no SD card, but may not be powered OFF until power is disconnected (at which point holding the power button finally does perform a poweroff).
Result: This is confusing and may drain the battery if power is then disconnected.
Note: I suspect this is related to the above issue, and is probably some sort of desync as well, if it's not to do with how the Pi handles "off" vs "no boot device."

4:) On some occasions (I can't confirm this for certain, or at least it's not reliably acting unusally) if power is applied via the Pi's USB port and it DOES have an SD card, the Hat automatically powers on (apparently by notification via the driver). However, it does NOT enable the halt/poweroff or hard shutdown features (holding the power button with the USB supply then disconnected still won't turn off power or shut down the Pi) and seems to be in a zombie state. Any press of the power button does seem to wake it, though, so the next attempt to shut it down works.
Result: It may worry the user or drain the battery.
Result: Otherwise as expected, aside from the initial zombie state. Assuming that this is real.

4b:) If the SD card does not have the drivers installed for the PiJuice Hat, it does not wake automatically and no UPS features are enabled. Manual Hat poweron (even after Pi boot/USB power application) and hard poweroff features work, though, even if USB power is still present (the Hat no longer supplies power, even if it's otherwise still powered).
Results: This is actually as expected (and kudos for making sure that this fails safely/works too!)

shawaj, 2020-02-07:
Have you got the pogo pin installed? And do you know which firmware version is on the PiJuice?

As this may affect our answers.

By the sounds of what you've asked, it sounds like your are running an older firmware without the pogo pin installed.

rdragonrydr, 2020-02-10:
It is using 1.3, which it says via the app is most recent.

I was also confused somewhat by the instructions, which I _think_ stated that the pogo pin wasn't needed with the version of Raspberry Pi (3B, original not plus) I had. Was I wrong?

shawaj, 2020-02-11:
Basically, when you "shutdown" a Raspberry Pi it goes into a halt state but it is still on. The only ways to restart it is to short the run pin to ground or to remove power and then reapply power.

With the PiJuice, if you are powering via the included PiJuice micro USB port, the board can remove power because all power is being managed by the PiJuice.

If you are powering the board via the Raspberry Pi micro USB, the PiJuice has no way to remove power from the Raspberry Pi. So the only way these functions will work is with the pogo pin as it gives the PiJuice the ability to short the run pin to ground.

We do have a way around this in future firmware but it is not available yet.

So my recommendation would be to power via the PiJuice micro USB or to solder the pogo pin in place.

tvoverbeek, 2020-02-11:
@shawaj Have to correct you on this.
From the V1.3_2019_02_28 release notes:
`-Added alternative wakeup via Rpi SCL (GPIO3) if run pin is not connected via pogo pin.`
From the Hardware README:
```
Note: As of Firmware v1.3_2019_01_15 pogo pin is no longer required to wakeup the
Raspberry Pi when powering from the Pi micro USB. A single press of SW1 on the
PiJuice will wakeup the Pi or a short of GPIO pins 5 & 6 will also wakeup the Pi.
This method has been introduced to accommodate the new Raspberry Pi boards
and any future boards.
```

shawaj, 2020-02-11:
@tvoverbeek oops! My bad there. I thought that was in firmware v1.4.

Apologies @rdragonrydr for the confusion.

tvoverbeek, 2020-02-11:
@shawaj The GPIO3 reset in the 1.3 firmware has a bug. It does not reset a number of internal flags after wakeup as the other 2 wakeup cases (power interruption, pogo pin).
This is fixed in 1.4.

rdragonrydr, 2020-02-12:
Is 1.4 available yet, or is it still in beta? But internal flags was actually my guess.

tvoverbeek, 2020-02-12:
@rdragonrydr Working on building the 1.6 packages right now.
If you want you can find the V1.4 firmware (part of the pijuice-base 1.6 package) here:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary
Use at your own risk. Not all 1,4 features are supported in the current software.
For changelog see: https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog the first section is about the firmware.