PiSupply/PiJuice #463: python3 pijuice service cpu consumption: how to configure?

brettbeeson, 2020-02-19:
Hi! Thanks for your great product! 

I'm using it with a solar panel, battery and pi zero. I notice the python3 pijuice service is running a lot (~6% CPU). I assume this is the service polls the hardware?

Can I configure this to poll less frequently? 

Thanks, Brett

tvoverbeek, 2020-02-19:
@brettbeeson Look at the dopoll loop in pijuice_sys.py:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py lines 323-343
Button events are polled every second, other stuff every 5 seconds.
You can increase the sleep (line 343) at the expense of a slower reaction time from the service.

brettbeeson, 2020-02-19:
Thanks Ton, I'll test that. Sounds like it matches my 'top' results.

brettbeeson, 2020-04-22:
That works fine.

closed 2020-04-22 by brettbeeson