PiSupply/PiJuice #465: PiJuiceHat, Buttons not responding and Crashing of Pi

Deelirium, 2020-02-20:
Hello,

I have a Pi Juice hat with a raspberry pi 4 running the latest image of raspbian.

Its buttons are unresponsive to every command except hold down for factory reset, this is despite the fact i have the software install, it is running under the correct users and i can see all the other stats of the battery/config.

I also have the issue of if i set the shutdown at loss of power it will randomly crash the raspberry pi, requiring me to then disconnect the power and reconnect to reboot. It does however correctly do a complete shut down and power off if i do pull the power. This occurs when powering the raspberry pi through both the USB C jack and directly through GPIO pins with a 5V 6A supply.

I have updated the firmware to the latest 1.4 just today, that has not solved the ongoing issues. I have tried fresh installs, resetting to factory settings and removing the battery for 30+ seconds and readding. 

Do i just have a faulty juice hat?

Cheers,
Sam.

shawaj, 2020-02-21:
So you have the latest v1.6 software installed and v1.4 firmware?

Deelirium, 2020-02-21:
Yes, I do. Originally when I first received and set up the juice hat I was running a 1.3 variety of the firmware. As of yesterday I updated to 1.4 and the problem still persists.

shawaj, 2020-02-21:
@Deelirium what are the commands you are saying the PiJuice is unresponsive to?

Deelirium, 2020-02-21:
All button operations. So switch 1 does not power on, on single press, or power off after long press/hard power off after longer press. Setting these commands to sw 2 and 3 also does not have any response.

shawaj, 2020-02-21:
Are you sure that you're pressing then successfully? If the button works to reset to factory settings then it sounds as if perhaps they aren't being pressed successfully.

Could you try with the end of a pencil or something like that that is small enough to press them?

Do you have any other add on boards connected to the Pi at the same time?

Deelirium, 2020-02-21:
Certain I am pressing them, have tested with a multimeter and get 1.2v when not pressed and 0 when pressed. Have tried with and without other attachments.

Hamhouke, 2021-02-19:
I am having the same issue, except that SW1 will power on the unit with a short press. Otherwise, the buttons seem to have no effect at all. I have updated the software and configured the buttons as you have. I tried it with a current and updated Raspian and RetroPie image.  Same behavior in both images.

mmilann, 2021-02-19:
@Hamhouke What is your SW1 use case, and how did you configured it in Pijuice buttons config menu?