PiSupply/PiJuice #466: Has anyone used the PiJuice with a PiBorg Reverse board?

wingers999, 2020-02-22:
Has anyone used the PiJuice with a PiBorg Reverse board?  

If so did you manage to get enough power via VSYS or P3 to power PiBorg Reverse board and motors? or if not how did you get round it?

Thanks

shawaj, 2020-02-22:
@wingers999 do you mean the PicoBorg reverse? https://www.piborg.org/motor-control-1135/picoborgrev

wingers999, 2020-02-22:
I did - sorry now corrected

shawaj, 2020-02-22:
It looks like that board needs 6 to 10 volts and, depending on what motors you are using, up to 5A per channel (total 10A).

PiJuice probably not suitable for this purpose.

With a large battery like the 12000mAh, you might get away with powering some smallish motors but don't know the exact spec of the motor driver chips and whether they'll work with only 5v

wingers999, 2020-02-22:
Okay thanks, I did wonder that.  So better off using PiJuice just for powering my Pi, and then having a separate battery pack directly connected to the PicBorg to power it and the motors

shawaj, 2020-02-22:
Yes that will work perfectly. 

Alternatively, you could use their Thunderborg motor controller which can power the Pi also from the same batteries https://www.piborg.org/motor-control-1135/thunderborg

wingers999, 2020-02-22:
Yes I had seen that, and I also have their BattBorg which would do same with PicoBorg, but then would defeat the point of having the PiJuice which I love!

shawaj, 2020-02-22:
True, I definitely won't stop you using PiJuice 😛

Did you have any other questions at all? If not I'll go ahead and close this issue.

wingers999, 2020-02-22:
:) 

No that is all thanks, I will get back to building my robot now :)

[image]

shawaj, 2020-02-22:
Cool! Enjoy the build. 

And send us pictures on Twitter / Facebook when you're done 👌

closed 2020-02-22 by shawaj