PiSupply/PiJuice #467: J5 doesn't start the Pi!

brettbeeson, 2020-02-23:
I've connected a momentary switch to J5 on my PiJuice Zero. I'm pretty sure this should operate like SW1. Specifically, if I short the 2 pins at J5 for <800ms it should start the Pi. 

However, when I short them, I observe the green LED flash: short-dim, short-bright, off. And the Pi doesn't start. the SW1 button works fine.

Am I missing something? Thanks, Brett.

[ I see there are some of issues about SW1 and J5 - this is different I think.]

shawaj, 2020-02-23:
How have you connected it? Is it soldered?

brettbeeson, 2020-02-24:
Yes, soldered it to two wires which have a momentary switch.

shawaj, 2020-02-26:
Sorry I just realised you have a PiJuice Zero not a HAT.

See here:
https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md#zero

shawaj, 2020-02-26:
So basically if you set SW2 in the menu to have the same settings as SW1 here - https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#pijuice-hat-config-buttons-menu

Then it should work

shawaj, 2020-02-26:
@brettbeeson could you give this a try and let us know how you get on?

brettbeeson, 2020-03-05:
Will do. I'm out of town but will do on my return. PiJuice working well timelapsing in Italian Alps!

shawaj, 2020-03-11:
@brettbeeson great! Send us some pictures of the Alps

brettbeeson, 2020-04-22:
I've tested that (i.e setup SW2 the same as SW1) and it works fine. Thanks!

And here is [a picture](https://www.dropbox.com/s/rm7bqvi2p55ritb/2020-03-07%2011.32.15.jpg?dl=0) of the alps, and the [timelapse camera](https://www.dropbox.com/s/2t7qz20nnj5l7su/2020-04-23%2008.59.30.jpg?dl=0) with the PiJuice zero. 

In the second photo you can see three white connectors: SW2 connection, user switch and solar input.

closed 2020-04-22 by brettbeeson

shawaj, 2020-04-22:
@brettbeeson that's a nice little rig! Would you mind if we tweeted the picture? Do you have a Twitter handle we could tag in it too?

brettbeeson, 2020-04-22:
Go nuts (that's a 'yes'). No twitter, just a [website](http://brettbeeson.com.au/battery-pizero-w-pijuice-zero-and-a-camera/) with technical details.