PiSupply/PiJuice #469: Does adding a Push Button on J5 require a resistor?

BryantDavidDavis, 2020-02-26:
Hi, I would like to add a push button to j5 on my pijuice zero, and wonder if I need to add a resistor or can I just solder the button directly to the terminals?  Thank you very much in advance for any answers!

shawaj, 2020-02-26:
No you can just solder a button on wires, no need for a resistor

shawaj, 2020-02-26:
@ChristopherRush it might be worth highlighting this somewhere / doing a brief guide or video on the soldering of J5?

BryantDavidDavis, 2020-02-27:
> @ChristopherRush it might be worth highlighting this somewhere / doing a brief guide or video on the soldering of J5?

Thanks a lot for your really quick response, I really appreciate it. Since you mentioned it, perhaps a quick note about it in the Button section of the documentation or somewhere like that might be easy to add, and helpful :)

shawaj, 2020-02-27:
@BryantDavidDavis yep, agreed. Will leave this open actually to remind us and we will close it once added

commit 929ed096 (2020-10-24): Add info about resistor etc

Closes #469
closed 2020-10-24 by shawaj