PiSupply/PiJuice #470: DEFAULT/ DIP SWITCH Battery charge profile malfunction

Hungarpol, 2020-03-01:
Can not use the default battery profile preset for BPX7. It changes to custom settings after confirmation of settings (does not fill all the fields prior to confirmation either). I have noticed it while I tried to troubleshoot the PiJuice not charging. I have found the battery profile changed to custom, with a few bottom fields filled only.  
It was OK before. The only actions I did since then were to turn VSYS 0/500 using a python script. 
If I select BPX7 from the drop-down list the charging starts to function. DIP switch set properly.    
[image]

shawaj, 2020-03-01:
I'm confused because you said you can't select BP7X by then further down you say you've selected it?

Sorry if I'm being stupid, just don't quite understand.

Hungarpol, 2020-03-01:
I am sorry for the confusing text, I will try to clarify -

1. As far as I understand, the setting according to the DIP switch will be in use, if the DEFAULT option is active in the battery profile tab

2. I used to have factory settings here, (which includes the DEFAULT option I guess), but one day when my pijuice did not charge anymore I realized that instead of the DEFAULT option I have an almost empty CUSTOM config active

3. When I choose DEFAULT from the drop down list it allows to pick it, but when I confirm with the Apply button it changes back to the empty CUSTOM  data set (see attached picture).

4. I used BPX7 from the drop-down list to make my system work, but if i am right, now it is independent of the DIP switch status

tvoverbeek, 2020-03-04:
Have you tried a MCU reset?

> Dual long press of SW1 and SW2 for 20 seconds will reset PiJuice HAT configuration to default. This applies to the MCU configuration only.

After the 20 seconds the LEDs should be flashing red during the reset.
Afterwards the MCU should be back on the default profile.
Note: All other MCU settings (buttons, etc.) will revert to their initial settings.

shawaj, 2020-10-24:
@Hungarpol did you manage to fix this?