PiSupply/PiJuice #471: PiJuice Broken Board?

RumoOr, 2020-03-01:
Recently noticed strange behaviour. see images below.

Pijuice leds not working. Battery status not available.
[Image 1](https://i.imgur.com/yW1TM0t.jpg)

When i put my finger on the board (witout any pressure), leds shows status:
[Image 2](https://i.imgur.com/Tfizoe4.jpg)

broken pijuice board?

shawaj, 2020-03-01:
@RumoOr what battery are you using? As it doesn't show anything on the picture?

Sounds more like a loose connection to me than anything.

RumoOr, 2020-03-02:
> @RumoOr what battery are you using? As it doesn't show anything on the picture?
> 
> Sounds more like a loose connection to me than anything.

Same behavior with Motorola (Default) Batterie and 5000mAH Extra Akku. Tested with both.

tvoverbeek, 2020-03-04:
Smells like a broken board. Time to contact the supplier for a replacement ??

shawaj, 2020-03-11:
@tvoverbeek yeah if it's happening with multiple different batteries I would agree. Something's not right.

@RumoOr where did you buy it from? From Pi Supply directly or someone else?

Epposan, 2020-03-12:
Are you using an extension header? If so, please check that you have one with pins that are square (0.6mm x 0.6mm) and not rectangular (0.6mm x 0.4mm). With the rectangular pins, the PiJuice slides on and off too easy (leading to bad contacts) 

I know of one firm in the Netherlands (with an Australian fruit's name) that sells the rectangular ones (without warning that they don't work well with the PiJuice). Adafruits has discontinued the rectangular ones (stated on their website)

shawaj, 2020-03-12:
@Epposan it looks like @RumoOr is using a PiJuice HAT on top of a full size Pi. So both headers will be soldered in the factory so this is unlikely to be an issue.

@RumoOr can you confirm which Pi you're using?

RumoOr, 2020-03-13:
@Epposan  @shawaj Hey guys, i got a new PiJuice from the Webshop (max2play) i bought it from. The New one is working! I guess it really was an hardware issue. The old one also was branded by Pi Supply, so i guess it was the original one. I use it on a RP3 B.

closed 2020-03-13 by RumoOr