PiSupply/PiJuice #472: PiJuiceZero: Schedule Power On / Shutdown after 25minutes / Power on when plugged in

ragingwhisky, 2020-03-02:
Hello,

I've recently started experimenting with the Raspberry Pi Zero (Wifi) and managed to get hold of a PiJuiceZero w/ Battery. I'm inexperienced with how best to configure the PiJuice so help and suggestion very much welcome :) 

What is the best way to perform the following use-case:

1. Turn on the Raspberry Pi Zero when charge is applied
2. Turn off the Raspberry Pi Zero when charge no longer applied
3. Schedule wakeup (assuming this would be the alarm) at 1800 each day
4. Pi Remains on for 30minutes and then shutsdown, ready for either step 1. to occur or step 3. to occur once more.

Any help and guidance most appreciated!
Thanks,

shawaj, 2020-03-02:
This is all covered in the software readme.

Have a look here:
https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#system-task-menu

shawaj, 2020-03-02:
Also here - https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#wakeup-alarm-menu

ragingwhisky, 2020-03-02:
Thanks for the comments, will be more prescriptive in the ask!

Using just the CLI mode (headless)

1. Turn on the Raspberry Pi Zero when charge is applied
-  Assumption from myself is "System Task > Wakeup on charge" - Trigger %:1 would sort this requirement? 

2.Turn off the Raspberry Pi Zero when charge no longer applied
- Assumption from myself is "System Events > "No Power"  and to utilise one of the sys_functions however unsure which of the top 3 functions is best suited to perform a safe shutdown of the Pi and not hinder any 'wake up'

3. Schedule wakeup (assuming this would be the alarm) at 1800 each day
- Assuming this would be "Wakeup Alarm" with "Wakeup Enabled" > set the time e.g. 
Hour: 18 
Every Day: check
Minute: (x)
Minute: (0)
Set Alarm 

4. Pi Remains on for 30minutes and then shutsdown, ready for either step 1. to occur or step 3. to occur once more.
- Unsure how best to go about this one?

Help on any of the assumptions / item 4 would be much appreciated.
Thanks in advance.

tvoverbeek, 2020-03-03:
1. Correct assumption. Note you can also set the limit to 0% (requires pijuice_cli the latest 1.6 debian package)
2. Connect SYS_FUNC_HALT_POW_OFF to "No Power"
3. Note the RTC is kept in UTC. So you should convert your local time to UTC to set the alarm.
4. This you will have to do with a program/script of your own. Note that when shutting down from your program/script you should send the wakeup-on-charge limit to the PiJuice before the shutdown.
Use the SetWakeupOnCharge(percentage) function in Python.
The internal wakeup-on-charge limit gets cleared when the PiJuice starts the Pi. In case of 2) the SYS_FUNC_HALT_POW_OFF will do this for you.
Also enable the 'Software Halt Power Off' function, so the 5V gets removed and the PiJuice goes to low power mode.

There is a complication when your Pi is connected to the internet. With the default settings the Pi sees a DS1339 RTC and the timesyncd daemon will synchronise the RTC to the actual UTC time **and clear the alarm flag**.
To avoid this change the eeprom address to 0x52. The DS1339 driver will not be automatically loaded.
If you need the time synchronisation add the line `dtoverlay=i2c-rtc,ds1307` to /boot/config.txt.
The ds1307 is identical to the ds1339, but does not have an alarm function. Hence the timesyncd will not clear the alarm flag.

ragingwhisky, 2020-03-04:
> 1. Correct assumption. Note you can also set the limit to 0% (requires pijuice_cli the latest 1.6 debian package)
> 2. Connect SYS_FUNC_HALT_POW_OFF to "No Power"
> 3. Note the RTC is kept in UTC. So [...]

Thanks for responding - that was really helpful. 

Regarding Point 4. i'm not too familiar with python and the syntax config for the PiJuiceZero (as a disclaimer), so from your comment, a method might be on a system event of "no power" (external power has shut off so running just on battery)
> Have the Wakeup enabled on alarm "PiJuice boots the PiZero at 1800 each day"
> Set a User Function  to enact a script on "No Power" state (assumption this would be read automatically as the PiJuice would recognise on 'wakeup' that it's running on Battery and not USB/5V input?
> script would be "example".py 

# from the PiJuice/Software pages

import pijuice, time
####
####
pj = pijuice.PiJuice(1, 0x14)
pj.rtcAlarm.SetWakeupEnabled(True)
###
<script to keep Pi alive for 30minutes and then shutdown>
SetWakeupOnCharge(1) 
# set at end of script so that wakeup can occur again - but risk(?) if Pi tries to wake up and has no  charge at all, so may need to be an amount that could last the 30minutes

Feel sheepish in asking, but would it be possible to advise what a 'good' py script would be?

End result:
1. Pi Turns on when USB charge is applied
Use: "System Task > Wakeup on charge" - Trigger %:1 

2.Turn off the Raspberry Pi Zero when charge no longer applied
"System Events > "No Power" and SYS_FUNC_HALT_POW_OFF

3. Schedule wakeup at 1800 each day
"Wakeup Alarm" with "Wakeup Enabled" > set the time e.g.
Hour: 18
Every Day: check
Minute: (x)
Minute: (0)
Set Alarm

4. Keep Pi on for 30minutes after wakeupalarm triggered each day, where after 30minutes Pi shutsdown and then the cycle repeats itself the next day at 1800 (unless the Pi is currently being given USB/5V power.

Thanks in advance and if helpful happy to IM.

tvoverbeek, 2020-03-04:
Questions: 
- When the Pi starts because it gets power (charge is applied) should it also run for 30 minutes, or should it run till power goes off?
- For the 30 minutes run. Should it shut down after 30 minutes elapsed time, irrespective of what it is doing? Or do you execute a task which takes about 30 minutes.
- When the Pi is running is it connected to the internet?

With the answers I probably could propose some example python scripts.

ragingwhisky, 2020-03-04:
> When the Pi starts because it gets power (charge is applied) should it also run for 30 minutes, or should it run till power goes off?

Run until power goes off when connected to working USB/5V supply.
- Remain **on completely** (irrespective of anything - Pi will be in active use when power is being supplied) until such a time when the USB/5V feed is turned off, at which point the scenarios kick in for wakeup if its 1800 / remain on for 30min / shutdown

> For the 30 minutes run. Should it shut down after 30 minutes elapsed time, irrespective of what it is doing? Or do you execute a task which takes about 30 minutes.

Definitely shutdown - no other additional tasks. 
- the Pi would be shifting log/monitoring outputs and have completed well before the 30minutes, however providing contingency in this regard - there will be no issue if the Pi shuts down and a transfer was occurring, as the log files will just transfer upon next batch run.

> When the Pi is running is it connected to the internet?

- To date, no internet connectivity as has been using private/segregated (airgapped) logging servers/wifi - this is just by coincident and wasn't by any master design.
- However may be safest/simplest to assume that the Pi may have line of sight to the internet in the coming months (so may risk the time sync issue between the Pi / Internet / PiJuice).

> With the answers I probably could propose some example python scripts.

Much appreciated again

angryfrenchguy, 2020-03-05:
Just to let you know, @ragingwhisk, you are not alone as I have been looking fo the same answers. The readme *is not* super clear. Many thanks to @tvoverbeek for his thorough answers. 

My use case is almost identical, with the pi connected to wifi. 

Speaking of not super clear : 

>When setting the Wakeup alarm for a repeated wakeup, after the initial reboot the Wakeup enabled capability is disabled due to the Raspbian RTC clock initialisation resetting the bit in the PiJuice firmware. To overcome this you will need to run a script to re-enable the wakeup-enable capability. The script is located in PiJuice > Software > Test > wakeup_enable.py

Just to make sure I understand. This means you can't program a recurring wakeup without this script? I guess I don't understand why that wouldn't be baked in the code, as it is a more common use case than a one time event.

tvoverbeek, 2020-03-05:
@ragingwhisky Here a setup which does what you want (I hope, have not tested everything).

- To avoid interference with the alarm flag by timesyncd:
  Set PiJuice eeprom address to 0x52.
  The device-tree blob in the eprom will then not be read and the
  RTC driver for the DS1339 will not be loaded.

- If you want the automatic time synchronisation by timesyncd add the line
  `dtoverlay=i2c-rtc,ds1307` to /boot/config.txt. This will load the DS1307
  driver at boot. DS1307 does not have an alarm function, hence the wakeup
  enabled flag in the (simulated) DS1339 will not be touched by the OS.

- If you run stand-alone (no internet) you will have to set the RTC to the
  current time at least once. Set the time first in Raspbian with the date
  command and then copy it to the RTC, either via the PiJuice GUI/CLI or with the
  command `sudo hwclock --systohc`.
  Note this sets the RTC time in UTC (same as when done by timesyncd).
  In the scripts below we only rely on the PiJuice RTC. With no internet the
  os time will be off.

- For your alarm time you then have to use UTC time. For example in continental
  Europe (UTC+1 when on standard time) you have to set the RTC alarm to
  17:00 to get the alarm at 18:00 local time. For the US east coast
  (UTC-5) you will have to set the RTC alarm to 23:00.

- Enable the "Software Halt Power Off" (and the "System Task") so when
  the shutdown is started by one of the scripts below, the PiJuice will
  remove the 5V after the shutdown and go to low power mode in case we
  are running on battery.

- On the System Events tab enable 'No power' and connect it to e.g. USER_FUNC1
  On the User Scripts tab define USER_FUNC_1 as /home/pi/myshutdown.py (or
  wherever my shutdown.py is located).

Script to run at boot: run30minutes.py
```
#!/usr/bin/python3

from pijuice import PiJuice
from time import sleep
import os

pj=PiJuice(1,0x14)

pjOK=False
while pjOK == False:
   stat=pj.status.GetStatus()
   if stat['error'] == 'NO_ERROR':
      pjOK = True
   else:
      sleep(0.1)
 
data = stat['data']
if data['powerInput'] == "NOT_PRESENT" and data['powerInput5vIo'] == 'NOT_PRESENT':
   # We are on battery, assume we need to shutdown after 30 minutes
   sleep(1800)
   # Before shutting down make sure wakeup_enabled and wakeup_on_charge have
   # the correct values
   pj.rtcAlarm.SetWakeupEnabled(True)
   pj.SetWakeUpOnCharge(0)
   os.system("sudo poweroff")
```
Call this script from /etc/rc.local.
Add this line to /etc/rc.local (assuming the script is located in /home/pi):
`python3 /home/pi/run30minutes.py &`
or if the script is executable:
`/home/pi/run30minutes.py &`

Script to run at 'no power': mypoweroff.py
```
#!/usr/bin/python3

from pijuice import PiJuise
import os

pj = Pijuice(1,0x14)

# Set wakeup_enabled and wakeup_on_charge just to be sure
pj.rtcAlarm.SetWakeupEnabled(True)
pj.SetWakeUpOnCharge(0)
# Now shut down
os.system("sudo poweroff")
```
Let us know how it goes.

ragingwhisky, 2020-03-05:
> @ragingwhisky Here a setup which does what you want (I hope, have not tested everything).
>
> * To avoid interference with the alarm flag by timesyncd:
>   Set PiJuice eeprom address to 0x52.
>   The [...]

Really appreciate it - will have access to the Pi setup tomorrow evening (UK local) so will report back on how I get on (good bad or ugly).

Thanks for helping on this

ragingwhisky, 2020-03-06:
Alarm: wakeup enabled for 1800 

PiJuice_cli - general > eeprom address set to 52

dtoverlay=i2c-rtc,ds1307 added as a single line to /boot/config.txt

System Task > 'system task' enabled + Software Halt Power Off + wakeup on charge (1%)

System Events > 'No power' Option set to USER_FUNC1
- User_func1 set to run /home/pi/mypoweroff.py

For 30minute "keep alive" before shutdown when on Battery:
- within /etc/rc.local added the line: python3 /home/pi/run30minutes.py &

Notes: 
Amended the poweroff.py - "juise" vs "juice", otherwise kept your script content.

Will set it going just now (will lose access to it shortly so next change will be to see what occurs for 1800 tomorrow.

Fingers crossed.

ragingwhisky, 2020-03-07:
Update: Wakeup on charge is working - looks like issue was the power input source (laptop) - however testing it with a standard Anker Powerbank and turned on as exdpected. 

User error.

Back with the Pi again and letting it charge before testing the "its off power - shutdown graceful until 1800 kicks by"

tvoverbeek, 2020-03-07:
Thanks for catching the typo in run30minutes.py. Corrected in my comment.

tvoverbeek, 2020-03-07:
Found some more typos/errors in myshutdown.py.
Here is a corrected version:
```
#!/usr/bin/python3

from pijuice import PiJuice
import os

pj = PiJuice(1,0x14)

# Set wakeup_enabled and wakeup_on_charge just to be sure
pj.rtcAlarm.SetWakeupEnabled(True)
pj.power.SetWakeUpOnCharge(0)
# Now shut down
os.system("sudo poweroff")
```

ragingwhisky, 2020-03-08:
Hi Ton,

Updated the scripts in /home/pi 
- Amended the previous "mypoweroff.py" to myshutdown.py for consistency with yourself (user_func script point to /home/pi/myshutdown.py)
- Updated the run30minutes.py also.

Just to check, within rc.local and config.txt that i've added the lines within the files correctly:
- dtoverlay=i2c-rtc,ds1307 added as a single line to /boot/config.txt added the first active (uncommented) line in the file 
- /etc/rc.local added the line: python3 /home/pi/run30minutes.py & added as first active (uncommented) line in the file

tvoverbeek, 2020-03-08:
@ragingwhisky Yes, your changes to /boot/config.txt and /etc/rc.local look fine.
I did some testing with a different schedule (not wakeup once a day and run for 30 minutes, but wakeup
every 10 minutes and run for 4 minutes) and that works.This is how I found the errors in the scripts.
So I am pretty optimistic that it should work for you.
Waiting for an update later today or tomorrow.

ragingwhisky, 2020-03-09:
All set up as expected - will find out in little over an hour whether i've done the config correctly! 

For reference (and to aid anyone else doing similar) here is the basic config i've got running.

RaspPi Zero w/ Wifi using PiJuiceZero Hat with battery attached to 3pin. 
Note: PiJuice is headless so all config done via command-line

Ton created 'myshutdown.py' with contents 

```
#!/usr/bin/python3

from pijuice import PiJuice
import os

pj = PiJuice(1,0x14)     # <-----

# Set wakeup_enabled and wakeup_on_charge just to be sure
pj.rtcAlarm.SetWakeupEnabled(True)
pj.power.SetWakeUpOnCharge(0)      # <-----
# Now shut down
os.system("sudo poweroff")
```

Ton create run30minute.py with contents:

```
#!/usr/bin/python3

from pijuice import PiJuice
from time import sleep
import os

pj=PiJuice(1,0x14)

pjOK=False
while pjOK == False:
   stat=pj.status.GetStatus()
   if stat['error'] == 'NO_ERROR':
      pjOK = True
   else:
      sleep(0.1)
 
data = stat['data']
if data['powerInput'] == "NOT_PRESENT" and data['powerInput5vIo'] == 'NOT_PRESENT':
   # We are on battery, assume we need to shutdown after 30 minutes
   sleep(1800)
   # Before shutting down make sure wakeup_enabled and wakeup_on_charge have
   # the correct values
   pj.rtcAlarm.SetWakeupEnabled(True)
   pj.power.SetWakeUpOnCharge(0)    # <----
   os.system("sudo poweroff")
```

From a Windows box, utilised WinSCP to copy the two scripts to /home/pi/ 
Within the PuTTy SSH session, 

Edited rc.local ( /etc/rc.local) and added the following line (as first active (uncommented) line in the file:
> 'python3_ /home/pi/run30minutes.py &' 

Edited /boot/config.txt (as first active (uncommented) line in the file:
> 'dtoverlay=i2c-rtc,ds1307' 

Whilst running under username Pi, ran PiJuice_cli (assume all other settings default):
Firmware: Update to latest
General
I2C address: 14
I2C address: RTC 68
GPIO Input Enable: Check
EEPROM write unprotect: <not checked>
Charging enabled: Check 
No battery turn on: <not checked>
Run pin: Not installed
EEPROM Address: 52
Inputs precedence: 5v_GPIO
USB micro current limit: 2.5A
USB micro in dpm: 4.2v
Power regulator mode: power_source_detection

Wakeup Alarm:
Status: OK
UTC Time: Matches current local
Wakeup enabled: Check
Day: X
Every day: X
Hour: 18
Minute: X
Set Alarm

System Task:
Statem task enabled: Check
Wakeup on Charge: 1
Software Halt Power Off: X / Delay Period 10

System events:
No power: User_Func1

User Scripts:
USER FUNC1: /home/pi/myshutdown.py

Hopefully that captures everything, however if i've missed anything major off just let me know.

ragingwhisky, 2020-03-09:
@tvoverbeek  
Update: didn't turn on at 1800 as expected.
Battery at 93% charge, so think mis-configuration on my part is the culprit.

Believe it might be a 'time' misconfiguration between the UTC/RTC - when viewing 'date' on the Pi = accurate to UK time, when viewing the time in Pijuice_cli = accurate to UK time.
-> Selecting "set RTC time" however and it syncs back 12 hours

For cleanliness, ran a 'reboot' however curious if this might be my downfall

tvoverbeek, 2020-03-09:
Smells you use am/pm. The 'Set RTC Time' copies  RPi time (UTC) to the RTC.
Since you are in the UK, right now your local time == UTC.
Do you use 24 hour notation on the Pi? Correct time zone on the Pi?

ragingwhisky, 2020-03-09:
24h clock wherever possible, however believe you're correct.

Will be able to log back in and check all the Pi time config in little over an hour so will report back.

ragingwhisky, 2020-03-10:
Manually set the localisation options within the raspi-config to utilise en_gb UTF-8

Updated local time (ntpd -q -g)

Then within the pijuice_cli - wakeup alarm - 'set rtc'

Confirmed with 'date' the local gmt, utc and rtc matches up (rtc few seconds out but not a dealbreaker).

ragingwhisky, 2020-03-10:
Some observations this afternoon after initiating a shutdown of the Pi (PiJuice remained on ints its 'low power state)

1. Attaching the Pi to a USB power source = did not automatically turn on (regardless of source type). Currently the lowest I can configure within powrjuice_cli is system task\Wakeup on Charge\ 1% (unable to make it 0 within the cli menu, or even change the 1 digit itself)

With the USB power coming in to the Raspberry Pi itself (charging up the PiJuice via the shared pin) i'm curious if the wakeup on charge is looking at USB Micro power input or if its agnostic and happy power coming in via GPIO power?

Curious if wakeup on charge is looking at the Battery => 1%, or if its looking at "charged up by 1%" i.e. went from 7% to 8% = 1% charge before turning on?
 

2. Checking timedatectl status
Local time tue 2020-03-10 13:16:52 GMT
Universal time tue 2020-03-10 13:16:52 GMT
RTC time: Tue 2020-03-10 14:59:11 

Local/UTC time were when the Pi shutdown, however RTC time remains correct. Am I correct in understanding that so long as the RTC time remains correct the wakeup_alarm will function correctly? 
A **reboot** of the Pi (once powered up) immediately synced all times.

tvoverbeek, 2020-03-10:
1. No, wakeup-on-charge will not work if you are powering via the Pi Usb.
You need to press SW1 to start the Pi again.
Your wanted scenario will only work when powering via the PiJuice USB.
pijuice_cli limits the wakeup on charge between 1 and 100.

Wakeup on charge is looking at charge >= 1%
But if you override the cli setting (as in myshutdown.py) it will look at charge >= 0%

2. Yes, as long as the PiJuice has power (even in low power mode) the RTC will keep the time.
Without Internet the time at boot will continue from the time at the last shutdown (package fake-hwclock).
When Internet is present the first run of timesyncd just after boot will synchronise all the times.

ragingwhisky, 2020-03-10:
Thanks for confirming - will swap out the port the cable is coming in on to the PiJuice itself.

Turned on today at 1800 local as planned. 

Will give the Pi one last cycle (tomorrow at 1800) and report back - many thanks again for the guidance and help.

ragingwhisky, 2020-03-13:
All appears to be working and well

@tvoverbeek much appreciated for the assistance

shawaj, 2020-03-14:
Great, thanks for the update. Will close this issue now but feel free to open a new one if you need anything else.

closed 2020-03-14 by shawaj