PiSupply/PiJuice #473: PiJuice locked up - no amount of pressing of SW1 made any difference

mikcatta, 2020-03-05:
I thought I was going crazy, but possibly not. 

I tried to start the Pi by pressing Sw1 and nothing happened. Tried a long press (and a longer press) and still nothing. Eventually, I removed power to the PiJuice/Pi for 30 seconds, re-applied power and voila!

I am testing recovery scenarios for a unit we have in the field, whereby it has power (I believe), but the PiJuice is not starting. This is a real pain as it will need an engineer to go to site to possibly just reboot it, any other ideas most welcome?

shawaj, 2020-03-11:
Is this a PiJuice HAT or Zero?

Is the power plugged into the PiJuice microUSB or the one on the Raspberry Pi?

Have you updated to the latest software and firmware?

mikcatta, 2020-03-12:
Yes it was plugged into the PiJuice HAT and it says it's on the latest.

The one in the field, the engineer got to look at yesterday and the PiJuice battery had buckled. What temperature is the battery ok upto and can we cool the CPU with the PiJuice HAT in situ?

tvoverbeek, 2020-03-12:
To cool the CPU you could use the Pimoroni fan shim.
You might need an extension header to mount the HAT, but it might fit without an extension header.

mikcatta, 2020-03-12:
Can you tell me the temperature the PiJuice battery will be ok at?

Would a fan push the heat towards the PiJuice & it's battery?

tvoverbeek, 2020-03-13:
@mikcatta if you look at the battery profile temperature parameters, temperatures below the cold and above the hot temperature should be avoided. Range is typically 0-60 degC.

The purpose of the fan is to transport the heat away from the CPU.
Note that between the warm and hot temperatures charging is reduced. Outside the working range charging is disabled. Similar on the cold side.

To avoid overheating the BP7X battery you could consider replacing it with an external battery,
Especially with a Pi4 the BP7X gets easily over 40 degC since it is sitting on top of the CPU.

mikcatta, 2020-03-20:
Excellent thank you

closed 2020-03-20 by mikcatta