PiSupply/PiJuice #474: PiZero not Powering On at Specified Battery %

fictionalname, 2020-03-08:
Hi Folks,

I've got a bit of an issue with my PiJuice Zero on my PiZero, which is being powered by a solar panel.

The settings for the PiJuice are like this : 

System Task
[X] System task enabled  
[X] Watchdog                       Expire period [minutes]: 15
[X] Wakeup on charge         Trigger level [%]: 2 
[X] Min charge                     Threshold [%]: 1       
[ ] Min battery voltage      3.3    
[X] Software Halt Power      Delay period [seconds]: 10 

 System Events
[X] Low charge:              < SYS_FUNC_HALT         > 
[ ] Low battery voltage:       NO_FUNC     
[ ] No power:                  NO_FUNC 
[X] Watchdog reset:          < SYS_FUNC_REBOOT       > 
[X] Button power off:        < SYS_FUNC_HALT         >  
[ ] Forced power off:          NO_FUNC  
[ ] Forced sys power off:      NO_FUNC  

My (possibly incorrect) assumption was that this configuration would mean that as the battery was discharging and it got to 1% the device would gracefully shut down the attached PiZero.  I then thought, as the solar panel charged the battery, when it got back up to 2% the device would then power the PiZero back on to continue working.  This does not appear to be the case.  I have come home on some occasions and found the device off, and when I manually powered it back on, I found that the battery was up around 15%.

I have just updated to the newest firmware and I'm still getting the same issue.

Can you please let me know what settings I should choose to have this functionality.

Thanks for your help!

tvoverbeek, 2020-03-09:
@fictionalname I find your limits for low charge (1%) and wakeup on charge (2%) pretty extreme when using a solar array. Switch off also depends on the capacity and condition of the battery.
The PiJuice has an emergency power off when the battery cannot supply the current needed by the Pi.
Probably your PiZero was woken up, but the battery + array could not provide the needed current and hence it was immediately shut down again.
I would try with min charge threshold set at 10% and wakeup on charge at 20 or 25%.

fictionalname, 2020-03-10:
Hi,

Thanks for your feedback, that's great.  I shall try it with the settings you have suggested and see if I get better results.

Cheers.

closed 2020-03-10 by fictionalname