PiSupply/PiJuice #475: RTC in Raspbian Buster 2020-02-13

tcsmedes, 2020-03-09:
I recently installed my PiJuice on a Pi 3B+ with a fresh install of Raspbian Buster image 2020-02-13, and everything seems to work fine except the clock I don't have any other hats installed and as far as I can tell my EEPROM address is set to 0x50, and when I run i2cdetect -y 1 I get 68 instead of the UU and when I run the hwclock -r i get "hwclock: Cannot access the Hardware Clock via any known method"  I have removed the fake-hwclock and I know I can add an overlay in config.txt but the documentation indicates that is not necessary unless I am using EEPROM address 0x52.

Is there something I am missing?

tvoverbeek, 2020-03-09:
Obviously i2c communication seems to be working (provided you got both 0x14 and 0x68 in i2cdetect).
Does the /proc/device-tree/hat directory exist?
If so what does `cat /proc/device-tree/hat/product` say (expected "PiJuice HAT")
If this is OK then the eeprom has been read.
Did it work OK on an other Pi?

tcsmedes, 2020-03-09:
Yes I received both 0x14 and 0x68 when I run i2cdetect
and yes I get PiJuice HAT when I cat /proc/device-tree/hat/product
I never ran it on another PI, I took it out of the box and antstat bag for the first time last night.
If I add dtoverlay=i2c-rts,ds1339 to config.txt everything seems to work fine although I haven't tried to play with schedules or events yet.  Just wasn't expecting that step unless I chnaged the EEPOM address.

tvoverbeek, 2020-03-09:
Do not know if on later PiJuice HATs they did not put the device-tree blob for the ds1339 in the eeprom.
Anyway the (simulated) ds1339 has  a problem when the Pi is connected to the internet.
The timesyncd which regularly syncs the time to the RTC also clears the RTC alarm, since the alarm can also be controlled by the OS.
Better to use  `dtoverlay=i2c-rtc,ds1307`. The ds1307 behaves the same as the ds1339 w.r.t. time, but does not have an alarm function. Hence the OS will not touch the alarm in the simulated ds1339.
You  have then full control of the alarm for (repeated) wakeup.

tcsmedes, 2020-03-09:
Thank you for you quick responses. I will give that a try.

shawaj, 2020-10-24:
@tcsmedes did you get this working? Can we close the issue?

tcsmedes, 2020-10-24:
Yes please.

Thanks
-Tad Smedes
K8TCS
N-Scale

"...misunderstandings and neglect create more confusion in this world than
trickery and malice. At any rate, the last two are certainly much less
frequent."
-Johann Wolfgang von Goethe, The Sorrows of Young Werther - 1774

shawaj, 2020-10-24:
Great thanks

closed 2020-10-24 by shawaj