PiSupply/PiJuice #476: Bad Battery?

tcsmedes, 2020-03-10:
Yesterday I installed a PiJuice new out of the box with a BP7X battery onto an RPi3B+.   I removed the plastic strip from the battery and plugged a Pi Foundation power supply into the MicroUSB on the PiJuice.  I don't have anything else connected to the PI and it is connected wireless to the network.

I used this set up off an on for about 12 hrs through out the day but the battery never fully charged, the blue and green LED continued to blink all day and the GUI always showed 99%.  I shutdown the PI last night from the OS and disconnected the power before going to bed.  This morning there wasn't enough power left in the battery to power up the PI.

My questions are am I doing some thing wrong or should I have done something different during the shutdown?
Do I just have a bad battery and should go looking for a replacement?
Is this just the behavior/performance I should expect?

ChristopherRush, 2020-03-10:
You already opened a support ticket for this and I have responded.

tcsmedes, 2020-03-10:
My bad, I was not aware the the two were linked and that this was more community based.  I appreciate the quick response.

closed 2020-10-24 by shawaj